SuperHouse/esp-open-rtos #515: OTA from "expressif SDK " to "esp-open-rtos"

wishys, 2017-12-11:
we used the SDK  and its bootloader form expressif before ,Can we upgrade our firmware  to this version through OTA?

nochkin, 2017-12-15:
I'm not familiar with the original Espressif's OTA process well enough to confirm it 100%, but I don't think it's compatible with the new rboot API and structure.
From what I experimented long time ago (things may got changed since than), it required to have 2 binaries depending on which Flash section it will go. There are may be additional things required by rboot.

HomeACcessoryKid, 2018-01-07:
I'd like to expand a bit on this subject.
MaximKulkin and I are building [Homekit compatible accessories](https://github.com/HomeACcessoryKid/esp-homekit-demo) and the market of accessible hardware has a tendency to use 8Mbit flash.
I also believe that OTA if essential for real devices in the field.
So, investigating the rBoot feature IN esp-open-rtos I read that it only supports BIG_FLASH mode.
HOWEVER, that is not going to work on a 8Mbit flash.

How can we get introduced to make the two-file-in-1Mbyte concept to work for esp-open-rtos.
It seems to even be the default of rBoot...  Any original author still around?

TIA,
HacK

HomeACcessoryKid, 2018-01-21:
I've just started a repo that hopes to make a universal robust OTA for esp-open-rtos:
https://github.com/HomeACcessoryKid/ota
In real life, booting inside the first 1M is not an issue, but it is impossible to make an elegant Makefile to take care of the alternative ld file. Good thing is that will not be an issue with this universal OTA approach.

nochkin, 2018-01-22:
@HomeACcessoryKid,
FYI, the OTA example in "libesphttpd" does not need to use "alternative ld file".

HomeACcessoryKid, 2018-01-22:
I'm pretty sure it is using flash bigger than 8Mb=1MB.  That is how it can avoid changing ld file.
However, my use case targets the very common 1MB flash. I'm also sure that any medium or large project cannot combine with libesphttpd in SSL mode and fit in 1MB.
Your model is surely valid for some users, but I want a security model where there is no doubt about what came from where. Also in my model an IoT sensor should be able to collect firmware without any user interaction, e.g. every month. And it should work for many thousands of devices... my home kit strategy will mean that it will be very accessible to many people.
Anyway, thanks for your advice.

nochkin, 2018-01-23:
Fitting into 1MB Flash limit may be a challenge. If you manage to do it while keeping HTTP+TLS, that would be nice.
Good luck with your project.

HomeACcessoryKid, 2018-03-30:
for those interested to look at the result so far, it is quite useable but still not well documented yet.
(https://github.com/HomeACcessoryKid/ota)  
(https://github.com/HomeACcessoryKid/ota-demo)  
you should start with flashing the ota repository, and then observe what happens... it all bootstraps out of that...  it can take a few minutes to complete the first time so be patient

feedback appreciated