SuperHouse/esp-open-rtos #533: how can I perform custom uart0

ashareye, 2017-12-31:
Hi 

I want to write a special uart driver with some equipments. But I cannot find any way to change ISR of uart 0, just like lib-driver in the original esp-rtos-sdk.

Please give me some advice thank you!

Your sincerely

doragasu, 2018-01-02:
As a starting point, you can have a look to `extras/stdin_uart_interrupt/stdin_uart_interrupt.c`. It has code to configure, set an interrupt handler and enable the receive interrupt for UART 0.

ashareye, 2018-01-05:
thank you for reply, yes I saw the "extras/stdin_uart_interrupt/stdin_uart_interrupt.c", but when I try to turn off the all system printf function like [#400](https://github.com/SuperHouse/esp-open-rtos/issues/400), it didn't work, even I redefine "#define printf(...) ets_printf(__VA_ARGS__)" in plamform.h. how can I do then? Any suggestion is appreciated.

thanks again.

ashareye, 2018-01-05:
and running in AP mode with DHCP, when pc connect the AP-wifi,  it always printf "station: 68:db:ca:4c:e2:74 join/leave, AID = 1", but I can't find "station:" in any source code, how can I git rid of it ???

ashareye, 2018-01-05:
I found the way but not sure it always work.

1.   add `printf sdk_printf` & `ets_printf sdk_ets_printf` in lib/allsymbols.rename
2.   define sdk_printf & sdk_ets_printf func in main.c like this:
`int sdk_ets_printf(const char *format, ...) {
	return 0;
}`
`
int sdk_printf(const char *format, ...) {
	return 0;
}`
3. `#define printf sdk_printf ` in main.c if you want turn off printf at all

then it print nothing with uart

closed 2018-01-05 by ashareye