SuperHouse/esp-open-rtos #535: select() server crashes on send()

doragasu, 2018-01-01:
Recently I updated to latest esp-open-rtos (I was working with a version from july 2016), and after adapting my code to the changes, I found that my application crashes. Just after accepting a TCP connection, an unaligned memory access exception is raised. More on this exception later.

So I rescued the code of my `select()` echo server test to try finding out if the problem is with my code or with esp-open-rtos. And it also crashes, just when the `lwip_send() is invoked`.

This is the code: https://github.com/doragasu/esp-open-rtos-echo-server/blob/master/tcp-echo.c?ts=4

And this is the output:
```
SDK version:0.9.9
mode : sta([REMOVED])
add if0
Waiting for connection...
scandone
add 0
aid 2
cnt 

connected with IORI_WIFI, channel 1
dhcp client start...
ip:192.168.1.132,mask:255.255.255.0,gw:192.168.1.5
Created listener socket number 3.
Listening to port 7.

connected with IORI_WIFI, channel 1
dhcp client start...
ip:192.168.1.132,mask:255.255.255.0,gw:192.168.1.5
Created listener socket number 3.
Listening to port 7.
Established connection from 192.168.1.10 on socket 4!
Data available on socket 4
Received 2 bytes

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
18:fe:34:d1:7b:06
load 0x3ffe8000, len 772, room 16 
tail 4
chksum 0xb3
load 0x40100000, len 2292, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
Wbl�|�b��n�p��nn�lnn�n�pn~�rb�n|bpbll`SDK version:0.9.9
mode : sta([REMOVED])
add if0
Waiting for connection...
scandone
add 0
aid 2
cnt 

connected with IORI_WIFI, channel 1
dhcp client start...
ip:192.168.1.132,mask:255.255.255.0,gw:192.168.1.5
Created listener socket number 3.
Listening to port 7.
```
Reset cause is 2, so an exception has occurred. Unfortunately, I don't know why, but no crash dump information is printed.

In my application, crash dump information was generated, but maybe the crash is not related to this one, because it did not occur on send(), it occurred just after accepting the connection. The crash dump pointed to `xQueueGenericSend()`, and the stack dump pointed to `_read_r()` <-- `xthal_get_ccompare()` <-- `my_task()`.

Maybe I am missing something, but this looks like a bug. Can anyone confirm this? Any suggestions about where to continue debugging?

NOTE that my code redirects stdout to UART 1 because I use UART 0 for other purpuses, but I have tested also removed the redirection code to use UART 0, and the crash happens the same.

ourairquality, 2018-01-01:
Could you please try increasing the echo_tsk stack. See a task stack overflow here running the example and increasing to 384 words seems adequate.

doragasu, 2018-01-01:
Damn, you are right, increased stack length and now it is working!

Time to test my app to see if it's the same problem. Maybe stack usage has increased since the last time I updated esp-open-rtos, because this worked fine before.

doragasu, 2018-01-01:
As I suspected, my problems were not related, because doubling stack sizes did not help to get my application working.

There is something that looks a bit strange: the stack dump, because it has the a5's just from the begginning:

```Fatal exception (9):
epc1=0x4020390f

0x4020390f: xQueueGenericSend at /home/jalon/src/esp8266/esp-open-rtos/FreeRTOS/Source/queue.c:2332

epc2=0x00000000
epc3=0x40204225

0x40204225: prvIdleTask at /home/jalon/src/esp8266/esp-open-rtos/FreeRTOS/Source/tasks.c:4662

excvaddr=0x3fff5239
depc=0x00000000
excsave1=0x4020390c

0x4020390c: xQueueGenericSend at /home/jalon/src/esp8266/esp-open-rtos/FreeRTOS/Source/queue.c:2332

Registers:
a0 4020390c a1 3fffbf70 a2  3fff1034 a3  00000001

0x4020390c: xQueueGenericSend at /home/jalon/src/esp8266/esp-open-rtos/FreeRTOS/Source/queue.c:2332

a4  00000020 a5  00000000 a6  3fff1fc0 a7  00000005
a8  00000003 a9  00000000 a10 00000000 a11 00000000
a12 3fff5201 a13 00000000 SAR 00000000

Stack: SP=0x3fffbf70
0x3fffbf70: a5a5a5a5 a5a5a5a5 a5a5a5a5 3fff1a1c
0x3fffbf80: ffffffff 3fffbfb0 3fff251c 402100c9

0x402100c9: _read_r at /home/jalon/src/esp8266/esp-open-rtos/core/newlib_syscalls.c:202

0x3fffbf90: a5a5a5a5 a5a5a5a5 a5a5a5a5 3fff1a1c
0x3fffbfa0: 00000005 4021c270 3fff251c 402141cc

0x4021c270: xthal_get_ccompare at ??:?

0x402141cc: LsdRecvTsk at /home/jalon/src/github/mw-fw-rtos/lsd.c:217

0x3fffbfb0: 00000005 3fff1fc0 0000007e 00000000
0x3fffbfc0: 3fff1848 00000000 00000000 00000000
0x3fffbfd0: 00000000 00000000 00000000 00000000
0x3fffbfe0: 00000000 00000000 00000000 00000000

Free Heap: 13220
_heap_start 0x3fff2860 brk 0x3fffcfe0 supervisor sp 0x40000000 sp-brk 12320 bytes
```

Could it mean that there is a stack overflow? (even though I have doubled stack sizes)

By the way, how did you manage to see the stack overflow? I did not get any information when the select server crashed.

ourairquality, 2018-01-02:
The value 0xa5 is the stack fill value, but that does not show the stack has overflowed. The stack overflow check occurs on task switches, so there is some luck in it being caught and it can be missed or a wdt reset might occur etc. The faulting address seems to be `excvaddr=0x3fff5239`, and this appears to be on the malloc heap, and any idea where this is? Unfortunately not many clues there. You could try enabling the lwip debug code to see if it detects any errors, so uncomment `#define LWIP_DEBUG` in lwipopts.h

doragasu, 2018-01-02:
Again, thanks for the tip!

Unfortunately, I have enabled most LWIP_DEBUG options and I cannot see anything suspicious, everything looks normal to me.

I am now suspecting `stdin_uart_interrupt` extra that I am using. I had a look to the sources, and it looks like they have not been updated to reflect all the changes that the `_read_r()` syscall has experienced (to support reads from stdin, lwip and filesystem). Unfortunately testing this is not as simple as removing `stdin_uart_interrupt` from the extras, since my code expects the read() calls to yield until data is available.

doragasu, 2018-01-02:
I have reverted all the changes to `_read_r()` and `_write_r()`:

- Set LWIP_SOCKET_OFFSET to 0
- Change `_read_r()` and `_write_r()` to only read/write to stdin/stdout.
- Change `_close_r()` always return -1 and set EBADF.

And guess what, everything works flawlessly now :).

As I am not happy with this solution (I like the work done to make file descriptors more Unix like), I will try to make this work without modifying esp-open-rtos when I get some time.

I'm closing the issue, thanks for the help!!!

ourairquality, 2018-01-02:
Thank you for the feedback and narrowing it down. But that does leave some unknown issue in there somewhere. Perhaps the lwip support for this was not as solid as needed. Could I ask if your app was using any other hooks into `_read_r()` and `_write_r()` that might have had issues such as a file system? 

I'll go over the lwip `select()` code paths again to see if something has been missed, and perhaps try writing some test code to try to tickle the problem.

doragasu, 2018-01-02:
This is the repository with the code: https://github.com/doragasu/mw-fw-rtos
Build instructions need updating though. Unfortunately I do not think you will be able to test the code without the additional hardware required.

I am not replacing `_read_r()` but I am in fact replacing `_write_r()`, see main.c: https://github.com/doragasu/mw-fw-rtos/blob/master/main.c?ts=4

I did it this way because this code was written before the stdio/sockets/filesystem descriptor namespace unification, and back then I decided to write to UART0 when `fd == 0`, and to write to UART1 when `fd == 1`. Not elegant but worked well. For the sockets, I only used the `lwip_XXX` functions to avoid problems.

For my `_write_r()` replacement to work as expected, I assume that lwip functions (e.g. `lwip_send()`) will never call it. But until I take more time to chase the problem, I do not know why this is happening.

doragasu, 2018-01-02:
In case you decide to try guessing what's happening, maybe the trace I took with lwip debug information might be useful (but I did not see anything strange): https://pastebin.com/YNxdPEaj