SuperHouse/esp-open-rtos #538: sdk_spi_flash_erase_sector + sdk_spi_flash_write doesn't work

zeroflag, 2018-01-12:
I use the combination of _sdk_spi_flash_erase_sector_ + _sdk_spi_flash_write_ to write out same data to the flash memory. But after reading it back the result still contain what it was already there previously. In other words it doesn't change the content of the sector.

Here is an example code that tries to update the 100th sector.

```c
char buffer[8] = "test1234";
sdk_spi_flash_erase_sector(100);
sdk_spi_flash_write(100 * 4096, (uint32_t*) &buffer, 8);
char result[8];
sdk_spi_flash_read(100 * 4096, (uint32_t*) &result, 8);
printf("result: %.8s", result);
```

zeroflag, 2018-01-13:
I just realized if the flash mode is set to QIO then it works but if it's DIO it doesn't.

doragasu, 2018-03-18:
How do you set QIO mode? I have several modules with ID C84016 (sdk_spi_flash_get_id() returns 1640C8, but I think this number must be byte swapped to get the correct ID). With these modules, everything works fine. But I am now using other modules with flash ID E04016 (different manufacturer) and flash does not work. I have tried using both sdk_spi_XXX functions, and spiflash_XXX functions without success. Maybe my problem is the same as yours.

doragasu, 2018-03-18:
OK, I found out that the way was easier than expected: using the download tool.

I tired first downloading my application with all the modes, and none worked. Then I noticed that the bootloader was run before my application, so flashed the bootloader with QIO mode and... voila! It works :)