SuperHouse/esp-open-rtos #539: Does the MQTT example : mqtt_client.c really works ?

vortex314, 2018-01-13:
I've tested the example a few times, and keep on getting errors :+1: 
I subscribed to "$SYS/#" to see some traffic. 
```
ESP-Open-SDK ver: 0.0.1 compiled @ Jan 13 2018 16:18:28
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(18:fe:34:f3:a1:7c)
add if0
mqtt_task: started
mqtt_task: (Re)connecting to MQTT server test.mosquitto.org ... error: -1
WiFi: connecting to WiFi
wifi_task: status = 1
scandone
wifi_task: status = 1
add 0
aid 3
cnt

connected with ######, channel 3
dhcp client start...
wifi_task: status = 1
ip:192.168.0.101,mask:255.255.255.0,gw:192.168.0.1
wifi_task: status = 5
WiFi: Connected
mqtt_task: started
mqtt_task: (Re)connecting to MQTT server test.mosquitto.org ... done
Send MQTT connect ... done
Received: $SYS/broker/version = mosquitto version 1.4.90
Received: $SYS/broker/timestamp = 2017-07-26 11:13:58+0100
Received: $SYS/broker/connection/ks.ral.me.rnic/state = 0
Received: $SYS/broker/connection/ks.ral.me.cosm-energy/state = 0
Received: $SYS/broker/connection/ks.ral.me.cosm-irc/state = 1
Received: $SYS/broker/connection/ks.ral.me.cosm-frl/state = 1
Received: $SYS/broker/connection/ks.ral.me.eclipse/state = 1
Received: $SYS/broker/connection/ks.ral.me.cosm-sys/state = 1
Received: $SYS/broker/connection/(none).test/state = 1
Received: $SYS/broker/connection/gateway.test.mosquitto.org/state = 1
beat
got message to publish
error while publishing message: -1
Received: $SYS/broker/connection/ip-172-30-2-200.defenditbridge/state = 1
Received: $SYS/broker/connection/mediabox.hivemq/state = 1
Received: $SYS/broker/connection/cinder.Public_Bridge/state = 1
Received: $SYS/broker/connection/winkekatze.winkekatz/state = 1
Received: $SYS/broker/connection/jrojoo-All-Series.test-mqtt-org/state = 0
beat
got message to publish
error while publishing message: -1
Received: $SYS/broker/connection/OpenWrt1504793280.test-mosquitto-org/state = 1
Received: $SYS/broker/connection/br-john-jane/state = 1
Received: $SYS/broker/connection/block_controller.bridge-01/state = 0
Received: $SYS/broker/connection/Salem2.Public_Bridge/state = 1
Received: $SYS/broker/connection/GOVERNMENT.FLIGHTS/state = 1
beat
got message to publish
error while publishing message: -1
Received: $SYS/broker/connection/do1.test/state = 1
Received: $SYS/broker/connection/MoonshotPi.TestMosquittoOrg/state = 1
Received: $SYS/broker/connection/cell_controller.bridge-01/state = 1
beat
got message to publish
error while publishing message: -1
beat
got message to publish
error while publishing message: -1
beat
Connection dropped, request restart
mqtt_task: started
mqtt_task: (Re)connecting to MQTT server test.mosquitto.org ... error: -1
mqtt_task: started
beat
mqtt_task: (Re)connecting to MQTT server test.mosquitto.org ... error: -1
mqtt_task: started
beat
mqtt_task: (Re)connecting to MQTT server test.mosquitto.org ... done
Send MQTT connect ... done
beat
got message to publish
error while publishing message: -1
beat
got message to publish
error while publishing message: -1
beat
got message to publish
error while publishing message: -1
beat
got message to publish
error while publishing message: -1
```

UncleRus, 2018-01-15:
Just tested it, and works fine:

```
ESP-Open-SDK ver: 0.0.1 compiled @ Jan 15 2018 20:01:26
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(60:01:94:0a:8a:5c)
add if0
mqtt_task: started
mqtt_task: (Re)connecting to MQTT server test.mosquitto.org ... error: -1
WiFi: connecting to WiFi
wifi_task: status = 1
scandone
wifi_task: status = 1
add 0
aid 2
cnt 

connected with URH, channel 4
dhcp client start...
wifi_task: status = 1
wifi_task: status = 1
ip:10.0.1.116,mask:255.255.255.0,gw:10.0.1.1
wifi_task: status = 5
WiFi: Connected
mqtt_task: started
mqtt_task: (Re)connecting to MQTT server test.mosquitto.org ... done
Send MQTT connect ... done
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
beat
got message to publish
```

vortex314, 2018-01-15:
hmmm, ok I cannot explain it. Maybe a bad wifi. Thanks for confirming.

closed 2018-01-15 by vortex314