SuperHouse/esp-open-rtos #541: Crash in AUTH_WPA_WPA2_PSK mode during UDP

trjohn, 2018-01-17:
Hi There,

I am observing an crash issue as below when the esp is configured as SOFTAP and when continous UDP packets are sent out from a task without any delay. Please refer the attachement below.

Any idea what could be the cause for this issue?

Fatal exception (28): 
epc1=0x40220c0c
epc2=0x00000000
epc3=0x4022300e
excvaddr=0x00000098
depc=0x00000000
excsave1=0x00000000
Registers:
a0 00000000 a1 3fff95e0 a2  00000000 a3  3ffead10
a4  00000000 a5  00000003 a6  00000000 a7  00000002
a8  00000001 a9  00000000 a10 3fff4e32 a11 00000010
a12 3ffead10 a13 3fff77d0 SAR 00000010

Stack: SP=0x3fff95e0
0x3fff95e0: 40223579 40223682 00000030 00000008
0x3fff95f0: d97fcf5c 20a20b97 d9d516a6 00000008
0x3fff9600: 3fff7788 40223011 3fff4e02 00080000
0x3fff9610: 00000000 3fff4478 3fff9a38 3fff4de0
0x3fff9620: 3fff15e4 3fff9a38 3fff15e4 00000010
0x3fff9630: 00000002 00000000 3fff4e26 3ffeaca4
0x3fff9640: 40205335 00000000 3fff7788 00000008
0x3fff9650: 4020c7c0 3fff7788 3fff4de0 4020c7c9

Free Heap: 28300
_heap_start 0x3fff1aa8 brk 0x3fffa378 supervisor sp 0x3ffffb00 sp-brk 22408 bytes
arena (total_size) 35024 fordblks (free_size) 5892 uordblocks (used_size) 29132

[access_point.zip](https://github.com/SuperHouse/esp-open-rtos/files/1638733/access_point.zip)

trjohn, 2018-01-17:
The crash is observed only during long run and Issue not seem to occur during AUTH_OPEN

ourairquality, 2018-01-17:
The heap is getting low "(free_size) 5892" so a guess would be that some allocation failed, and the NULL pointer was dereferenced giving the exception. Might want to check the strlen() of the buffer that is not null terminated, and threads can not return and to end call `vTaskDelete(NULL)`, perhaps look into lwip to see where the memory is going and if it can be managed better.