SuperHouse/esp-open-rtos #552: WiFi packet loss

trjohn, 2018-01-22:
Hi,

I am facing an issue when I send 10 UDP packets over wifi without any delay in between each send, the receiver is receiving just 6 packets. I ensured that 10 packets are sent to **sdk_ieee80211_output_pbuf** and just 6 packets are received from **sdk_ieee80211_deliver_data**.

But if I add a delay(30ms) between each send in the sender side, the receiver is receiving all 10 packets correctly.

Even though UDP is not correction oriented and the application has to take care of retransmission, the fact that always the wifi sdk is missing the packets is a question mark.  Does the wifi libraries in this sdk need some time interval between each send?

FYI, I tested this behavior on both 80Mhz and 160 Mhz. Issue is found on both the system clocks.

Anybody could help me to understand this issue better?

Thanks in advance.

ourairquality, 2018-01-22:
There are a limited number of buffers for output packets, these are internal structures like a pbuf, and there might be only 6 of this type. The return value of `sdk_ieee80211_output_pbuf` might be 0 on success and 1 if it fails, so perhaps you could use that to back off a little and retry.

Not sure how practical it would be to expand the pools.

Only other option might be to hack into `sdk_esf_buf_recycle` to keep track of the queue and schedule the next TX packet when there is room. This is probably called from the pp_task but might signal an app thread to send more packets. It seems to reference static data making this difficult but perhaps not impossible.

trjohn, 2018-01-23:
Hi 

You are correct. I could see lot of failures(returning 1) from sdk_ieee80211_output_pbuf when packets are sent without any delay. I will have to manage this from application side.

By the way, where can I find the source code for sdk_esf_buf_recycle and pp_task?

Thanks

ourairquality, 2018-01-23:
@trjohn There is no source code for these otherwise it would be a lot would easier to fix. Some progress has been made adding source code for sdk binary code, but not that far. Some of the hacks involve modifying the binary to call another function that we can implement in source code. Some parts I have tried replacing with upstream code, so replaced the libwpa, but it needs a lot more work. I find it hard to manually translate, I keep making endless mistakes and introduce even more bugs, and huge waste of time.

trjohn, 2018-01-23:
@ourairquality  thanks for your feedback. Just one query, why this sdk is not taking the latest official espressif libraries?

ourairquality, 2018-01-23:
@trjohn Licensing changed, so people are probably not keen to freely contribute under a license that restricts code to being used only on the esp8266 device. It's not trivial to swap in newer sdk binaries, and we have some source code and might have to abandon that - I did attempt this and found a number of issues and there was no help to resolve them, and personally might have done things differently.