SuperHouse/esp-open-rtos #560: Data Races

Zaltora, 2018-02-04:
Just a question. I got multiple global variable ( Read-only acces through a extern const pointer outside of the file) in my app.
It is possible to get data races problem with esp8266 (one core) ?
If yes, using  "Critical rtos API" can help ? 
bonus: esp32 can get this problem (2 core)?

Zaltora, 2018-02-09:
Sometimes i get random reboot on error. The cause of error is related to memory problem and I can not reproduce the error by doing the same things that led me to the error.

I am not out of memory:  RTOS heap  => +10kBytes
I was thinking of free/malloc problem, but after check i see nothing. Anyway when i can't reproduce. 
the random reboot rarely happens.
I suppose a memory alignment problem too but i don't no how check it.
I suppose a memory access between task problem so i begin to change my system and use local variable for treatments then write it in a global variable (read-only) with vEnterCritical() API but i don(t no if it is will got effect or it is useless.

flannelhead, 2018-02-09:
@Zaltora, critical section is of course an effective way of protecting from concurrent access. However that is also a bit heavy-handed approach. FreeRTOS also has a mutex API (see the FreeRTOS) docs, which you can even use from ISRs.

To your original question: yes, race conditions can be a problem even though we have only one core - after all there are multiple concurrent tasks running.

Zaltora, 2018-02-10:
Yeah forget about mutex, i will use them.
About theses global variable, this problem can appear if two task want read the memory ?. If no i guess i need protect the writing ?
Other situation:
- A lower priority task want read a variable through a pointer const. (Read-only location then)
- A higher task want run and write into this memory location.
- The lower task come back and will fail to read. (reboot)
I need to protect lower priority task when reading theses variables with a mutex ?