SuperHouse/esp-open-rtos #562: sntp_fun.c incorrecty computes fractional seconds

jeffsf, 2018-02-06:
Examination of the results of `sntp_get_rtc_time()` in operation shows both values over 1,000,000 as well as negative values. When using the SNTP module, this breaks other time-related utilities as well.

The issue appears to be `sntp_fun.c:84` which appears to fail to properly rescale the result by 4096. It appears that changing that line to 
```
		*us = ((base * cal) % (1000000U<<12)) >>12;
```
resolved the range issue and a quick test shows an advance of ~0.10 seconds with a task delay of 100 ms. 

Patch coming after I review the algorithm and confirm the correctness of the patch.

ourairquality, 2018-02-06:
Fwiw the rtc clock appears to drift too much to be usage as a source for ntp. Guess would be that the rtc clock was intended for a low power clock when sleeping. The sntp code should probably be rewritten to use another clock source, something that does not change when the cpu clock speed changes, such as the system clock.

jeffsf, 2018-02-06:
Agree that the "RTC" has both too much granularity (~25 ms on my chip) as well as short-term variability with `sdk_system_rtc_clock_cali_proc()` showing on the order of 1000-2000 ppm changes over even a minute to be a "good" source of time. Long-term drift, of course, is another issue. Even if it were used during deep sleep, I'd want to wait for or poll a server to get time on wake.

I do need a sense of real time to be able to time-tag MQTT messages with QoS of 1 or greater to one second or better, especially when they may be queued for later delivery during connectivity losses. Remote syslog is another place that coordinated time is important to me.

At least I can fix what's there and have something, as imperfect as it may be, to use for "real" time.

PR #563 (2018-02-09): Improve extras/sntp -- Issue #562
Primary change is correcting the missing scaling of the microseconds calculation 
Resolves #562 

Other changes address:
* Capture RTC count before any logging
* No "obligatory" printf() logging
* Change logged message to show:
  * adjustment about to be made in seconds
  * time since last SNTP update
  * ratio of the two as ppm
  * "cal" value (raw as int, as before)
* Initialize "cal" value from sdk_system_rtc_clock_cali_proc() 

I'm more than happy to squash this, or split into mutiple commits
closed 2018-02-09 by UncleRus