SuperHouse/esp-open-rtos #566: help wanted: Flashing goes wrong on Wemos Lolin32

bobmosh, 2018-02-12:
Hey lovely people,

I'm having a hard time flashing any code to my Wemos Lolin32. I have tried many different methods, and examples but I can't seem to get it working.

I've now gone down to just trying to flash the blink exam to the ESP32 and even that doesn't work.

I'm using the following commands:
```
Ferdinands-MacBook-Pro:blink Ferdi$ make all
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/ets_timer.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/misc.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/os_cpu_a.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/spi_flash.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/timers.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/uart.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/user_interface.c
AS /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libmain/xtensa_context.S
AR build/open_esplibs_libmain.a
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libnet80211/ieee80211_ets.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libnet80211/ieee80211_hostap.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libnet80211/ieee80211_input.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libnet80211/ieee80211_sta.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libnet80211/wl_cnx.c
AR build/open_esplibs_libnet80211.a
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libphy/phy.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libphy/phy_chip_v6.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libphy/phy_sleep.c
AR build/open_esplibs_libphy.a
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libpp/esf_buf.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libpp/if_hwctrl.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libpp/lmac.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libpp/pm.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libpp/pp.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libpp/wdev.c
AR build/open_esplibs_libpp.a
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libwpa/os_xtensa.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/open_esplibs/libwpa/wpa_main.c
AR build/open_esplibs_libwpa.a
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/examples/blink/blink.c
AR build/program.a
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/croutine.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/event_groups.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/list.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/queue.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/stream_buffer.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/tasks.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/timers.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c
AR build/freertos.a
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/esp_interface.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/sys_arch.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/api_lib.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/api_msg.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/err.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/if_api.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/netbuf.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/netdb.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/netifapi.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/sockets.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/api/tcpip.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/altcp.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/altcp_tcp.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/def.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/dns.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/inet_chksum.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/init.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/ip.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/mem.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/memp.c
CC /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/lwip/lwip/src/core/netif.c
[... 121 lines cut ...]
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 127.3 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 2707.6 kbit/s)...
Hash of data verified.
Compressed 253488 bytes to 185917...
Wrote 253488 bytes (185917 compressed) at 0x00002000 in 16.4 seconds (effective 123.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

From the outside it looks like everything works fine, but for example If I try to connect to read the print statements (not in this case, since it has none but in others) I get the following error:

´´´
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
´´´

Why is that? I can't seem to figure it out, so I thought I asked the people way smarter than me because I'm starting to pull my hair out on this one!

Is there someone here that might be able to help me figure this out?

flannelhead, 2018-02-12:
@bobmosh, please note that `esp-open-rtos` is for ESP8266 only. You seem to have ESP32 on the board.

bobmosh, 2018-02-12:
@flannelhead, oh man. I am so sorry about that stupid mistake and the issue I opened up for that!
I'll get my hands on an ESP8266 and close the issue for now.

Thanks for the hint!

closed 2018-02-12 by bobmosh

flannelhead, 2018-02-12:
@bobmosh, ESP32 is a way more powerful chip than the ESP8266. [`esp-idf`](https://github.com/espressif/esp-idf) is a programming environment similar to `esp-open-rtos` but for ESP32 (it's based on FreeRTOS). I bet you'll find that interesting.

bobmosh, 2018-02-12:
@flannelhead Thanks, I'll definitely give that a shot! I'm going for HomeKit compatibility, so I might not be able to use it in this special scenario, but it's definitely good to keep in mind! Thanks a ton!