SuperHouse/esp-open-rtos #568: Multi-channel software PWM

nochkin, 2018-02-15:
Hey, ESP8266 hackers.

I needed multi-channel PWM for my own project, but the PWM module included in esp-open-rtos does not support different duty for multiple channels.
I tried to [port](https://github.com/SuperHouse/esp-open-rtos/pull/522) someone's else library, but did not realize its license is not compatible with esp-open-rtos.

So I developed my own PWM library which supports multiple channels. The POC looks good and it appears to be working. It probably needs more testing for some crazy scenarios, but I think it's ready to be shared with the community.
The most important part is that I'm releasing it under MIT license for esp-open-rtos.

I picked the existing API from extras/pwm/pwm.h and used that, but practically I had to rewrite the whole pwm.c file while keeping pwm.h almost intact.
My new stuff is compatible with the original API, the original example works.

Here is my dilemma and choices on how I think it should be PR'ed:
1. Have the PR to replace the existing PWM library. The only problem is since my code is a complete rewrite, it may introduce unexpected behaviour/bugs to the projects which use the original PWM already.
2. Have the PR to create another PWM library (i.e. "extras/pwm2" or something like that). I'll probably have a different API to avoid confusion.

Let me know what you think.

flannelhead, 2018-02-15:
@nochkin this sounds interesting! Do you have the library already available somewhere where I could have a look at it?

I'm especially interested in what kind of resolutions and frequencies it can produce.

nochkin, 2018-02-15:
It's currently available at: https://github.com/nochkin/pwm

I need to do more technical-related tests using a logic analyzer. I'm sure the precision can be improved (i.e. considering overheads, etc).
What I've done is kinda basic, I needed it to control 2 motors at variable speeds.

nochkin, 2018-02-23:
Anyway, I decided to go for a new name (multipwm) to avoid possible issues if someone uses the original library.
I will prepare the correspondent PR shortly.

nochkin, 2018-02-24:
Merged as per PR #579.

closed 2018-02-24 by nochkin