SuperHouse/esp-open-rtos #572: Make a component can be use with esp-idf and esp-open-rtos

tuanpmt, 2018-02-17:
There are `component.mk` file for **esp-idf** component and **esp-open-rtos**. However, with **esp-open-rtos** there is an additional:  `$(eval $(call component_compile_rules,componet_name))`
Is there a simple way to check and compile a component for both?

Thanks

trustchk, 2018-03-03:
Hi there, I had some success with checking for that component_compile_rules symbol like so:
```makefile
ifdef component_compile_rules
# esp-open-rtos (ESP8266)
INC_DIRS += $(name_ROOT)/...
name_INC_DIR := $(mpd_ROOT)/glue
name_SRC_DIR := $(name_ROOT)/...
$(eval $(call component_compile_rules,name))

else
# esp-idf (ESP32)
#COMPONENT_ADD_INCLUDEDIRS := ...
#COMPONENT_PRIV_INCLUDEDIRS := glue
#COMPONENT_SRCDIRS := ...
endif
```
Basically, I looked for defines and exports that are set by only one of the frameworks and ended up using the most obvious one. I'm sure there are more elegant versions, but so far it worked for me. If this works for you and no one comes up with solutions, it might be worth documenting somewhere.

For esp-open-rtos, I added it the component folder to a "extras" subfolder inside my project folder, for esp-idf it went into the usual components folder.

tuanpmt, 2018-03-03:
thank you very much. This is exactly what I need

closed 2018-03-03 by tuanpmt