SuperHouse/esp-open-rtos #588: ESP-12F stuck.

orhaneee, 2018-03-08:
Hi everyone, thanks for your work firstly.

I have a ESP12-F module. I flashed some of the examples to try out but it always stops at
```sh
rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
```

I have changed parameters.mk for 4 MB flash size. I did not change common.mk by the way.

Thanks

flannelhead, 2018-03-08:
Hi, please try doing a full mass erase with `esptool`, then reuploading the firmware. That has worked for me before.

orhaneee, 2018-03-09:
Hi @flannelhead thanks for your answer but unfortunately it does not work :(

nochkin, 2018-03-09:
I would try to play with flash modes to see if you need DIO or DOUT set.

orhaneee, 2018-03-09:
@nochkin I was using QIO before and I have tried all of modes.. no success..

ourairquality, 2018-03-09:
You might need to trace how far it is getting to find some more clues. See core/led_debug.s So turn the led on/off as needed at some stage in the early boot, and keep moving it further in the source to locate where it is stopping. For a start try at the start of `sdk_user_start` in core/app_main.c

skcks, 2018-03-09:
try this:
`make flash FLASH_MODE=dout`

Zaltora, 2018-03-09:
I got some ESP-12F and work it is work with DIO mode (i don't no why is not work on QIO, because it is a QIO model). Anyway i will get soon ESP-12S and will do some test on it.
Take care about TX pin, when i design some PCB, i have detected this pin need to not be pulled to GND or you will get some boot problem.

chenli2015, 2018-03-17:
I meet the same problem!

Zaltora, 2018-03-27:
I was wrong !!
Never put TX pin to GND at start, esp8266 will never boot. I don't know why. It is can be a hardware problem.

orhaneee, 2018-03-30:
@chenli2015 thanks for your solution. I will try this as soon as possible and let you know. I also think that main cause might be rboot bootloader.

apiel, 2018-04-21:
Did you solve your issue?

chenli2015, 2018-04-23:
@apiel did you use the same flash tool like me? the problem is new flash partition type not support by "old" rboot.

apiel, 2018-04-23:
Yes, i was looking at rboot.c and it has the content as u mentioned in your previous message.

Thx, hopefully someone will find the solution :-)

orhaneee, 2018-04-23:
Hi again, sorry for late reply. I did what @chenli2015 said and it seems to be resolved.
Thanks again.
So, I am closing it.

closed 2018-04-23 by orhaneee

apiel, 2018-04-23:
I tried to do exactly what is mentioned by  @chenli2015 without success

I flash it with:
         esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 16m -fm dio -ff 40m 0x0 ../../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/blink.bin

[image]

Any more idea?

apiel, 2018-04-24:
> @apiel <https://github.com/apiel> Please try my config.bin setting:
> e1 01 00 00 00 01 00 00 00 00 20 00 ......(all 0s)
> Not flash blank_config at 0x1000

What do you means? How can I do this?

Thanks for your help ;-)

orhaneee, 2018-04-24:
@apiel download a hex editor (i.e GHex in Linux) and open config.bin file and change its contents as @chenli2015 says.

chenli2015, 2018-04-25:
@apiel @fvpalha open "bootloader" floder and you will see "firmware_prebuilt" just open it. there may has a file named "blank_config.bin". use the following setting:

ourairquality, 2018-04-25:
Submitted a PR for rboot https://github.com/raburton/rboot/pull/37 and a PR to update esp-open-rtos to the current rboot master https://github.com/SuperHouse/esp-open-rtos/pull/617 and a little more support in esp-open-rtos https://github.com/SuperHouse/esp-open-rtos/pull/616

Could I ask which flash tool is causing the problems, and does the current esptool not work for some configurations? Just not clear from the discussion why the modified config.bin is even necessary?

chenli2015, 2018-04-29:
@apiel No problem for all steps. Try the following steps to debug:
1. clear flash and ensure that it is really cleared!
2. flash rboot.bin, config.bin, firmware one by one and reset after each burn, watch the boot information.
3. ensure the firmware works