SuperHouse/esp-open-rtos #592: httpd webserver issue.

agmuntianu, 2018-03-13:


flannelhead, 2018-03-13:
Could you provide a patch that fixes the issue?

I'm asking because I've been (slowly) working on upstreaming the `httpd` module to lwIP. The `httpd` in this repo was taken from lwIP in the first place, and WebSocket support was added but never upstreamed. The `httpd` module in lwIP has evolved quite a bit after that and probably also this has been fixed there... If the `httpd` in the IDF has really been ported from the one in this repo, the situation really is more funny than I expected. My WIP upstreaming work is [here](https://github.com/espway/lwip) and the respective ticket [here](https://savannah.nongnu.org/patch/?9525). If someone wants to help with getting this upstreamed, it would ultimately be beneficial for both `esp-open-rtos` and the IDF.

agmuntianu, 2018-03-13:
Hi,

Attached a patch file.

[httpd_issue.patch.txt](https://github.com/SuperHouse/esp-open-rtos/files/1807230/httpd_issue.patch.txt)