SuperHouse/esp-open-rtos #606: vTaskDelete can't free memery

donny681, 2018-04-13:
I found that it delete the task successfully.But it can't free memery.It failed to xTaskCreate  the same task after delete the task. I think that it may be something wrong with the system.

ourairquality, 2018-04-13:
Could you provide an example to reproduce the problem?

joostn, 2018-05-10:
According to https://www.freertos.org/a00126.html :

NOTE: The idle task is responsible for freeing the RTOS kernel allocated memory from tasks that have been deleted. It is therefore important that the idle task is not starved of microcontroller processing time if your application makes any calls to vTaskDelete (). Memory allocated by the task code is not automatically freed, and should be freed before the task is deleted.

So it appears the memory is not released immediately after calling vTaskDelete. Could that be the issue?

ourairquality, 2018-05-12:
The idle task does run, and the memory does appear to be freed. There might be some delay, or the idle task might be starved and not running in some particular app, or the heap might become fragmented, but these do not appear to be a issues that needs to be addressed here. A simple test looping creating a task that deletes itself does not leak memory here.