SuperHouse/esp-open-rtos #619: Static IP in station mode

korstiaanS, 2018-04-27:
Hi,

Can somebody help me with some code how to use a static IP in a station mode?
I really need to use static IP's.

Thanks for wanting to help me.

Korstiaan

ourairquality, 2018-04-27:
The examples/wificfg/ includes support for a static station IP address, so that is one example see `wificfg_init()` in extras/wificfg/wificfg.c

korstiaanS, 2018-04-27:
Hi,
I tried this (same sequence) but still I get a DHCP address.
I can see this also in the serial monitor; there it says "dhcp client start..."
I used the following code:

```
static void wifi_init() {
    uint8_t macaddr[6];
    struct ip_info static_ip_info;
    struct sdk_station_config config;
    memset(&static_ip_info, 0x0, sizeof(static_ip_info));
    
    strcpy((char *)config.ssid, WIFI_SSID);
    strcpy((char *)config.password, WIFI_PASSWORD);
    config.bssid_set = 0;
    
    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_dhcpc_stop();
    
    IP4_ADDR(&static_ip_info.ip, 10, 0, 12, 200);
    IP4_ADDR(&static_ip_info.gw, 10, 0, 12, 1);
    IP4_ADDR(&static_ip_info.netmask, 255, 255, 255, 0);
    
    sdk_wifi_set_ip_info(STATION_IF, &static_ip_info);
    
    sdk_wifi_station_set_config(&config);
    
    sdk_wifi_get_macaddr(STATION_IF, macaddr);
    sprintf(responseprefix, "%02X%02X", macaddr[4], macaddr[5] );
}

```

This is called from user_init().

ourairquality, 2018-04-28:
Could you trace `sdk_eagle_auth_done()` in `open_esplibs/libwpa/wpa_main.c`. This is where the static IP address should be applied, and where the dhcp client is started. If it is printing "dhcp client start..." then the `sdk_dhcpc_flag` must not be `DHCP_STOPPED` which would be wrong. Then could you trace `sdk_wifi_station_dhcpc_stop()` in `libmain/user_interface.c` and perhaps you can spot what is going wrong.

korstiaanS, 2018-04-28:
I traced it and I see that in the call `sdk_wifi_station_dhcpc_stop()` I don't have `if (sdk_wifi_get_opmode() == 2)` and also the program does not enter `if (netif && sdk_dhcpc_flag == DHCP_STARTED)` but then I checked if the flag `if (sdk_dhcpc_flag == DHCP_STARTED)` and this is TRUE? So that means that the problem lies with netif?

ourairquality, 2018-04-28:
Thank you. Would the following change fix this?  So always set the `sdk_dhcpc_flag` to `DHCP_STOPPED` even if there is no `netif` when called.

```
bool sdk_wifi_station_dhcpc_stop(void) {
    struct netif *netif = _get_netif(0);
    if (sdk_wifi_get_opmode() == 2) {
        return false;
    }
    LOCK_TCPIP_CORE();
    if (netif && sdk_dhcpc_flag == DHCP_STARTED) {
        dhcp_stop(netif);
    }
    sdk_dhcpc_flag = DHCP_STOPPED;
    UNLOCK_TCPIP_CORE();
    return true;
}
```

ourairquality, 2018-04-30:
Thank you for testing. PR https://github.com/SuperHouse/esp-open-rtos/pull/621

closed 2018-04-30 by UncleRus