SuperHouse/esp-open-rtos #624: Deep Sleep Excessive Current Draw: ~300uA -- ESP8266-12E

brandonbraun653, 2018-05-14:
I'm working on a low power WiFi enabled alarm system for my apartment using the ESP8266-12E and am having some issues with the deep sleep current current draw. I am using the `sdk_system_deep_sleep()` function, but with a bare bones chips on the breadboard and using only the necessary pull up/down resistors for normal operation, I get just about 300 uA. This was measured using one of the EEV Blog uCurrent Gold meters as well as with a second handheld device. 

I've run into discussions on similar issues [here](https://github.com/esp8266/Arduino/issues/3979) and [here](https://github.com/nodemcu/nodemcu-firmware/issues/1625) but those are for different platforms, Arduino and NodeMCU respsectively.

Originally I was drawing near 2.2 mA in deep sleep, but discovered that the SPI Flash chip was still active. Adding a 10k pull-up on CS0 solved that quickly and brought me to the my current 300 uA value. Since most of the discussions I've read online about this kind of problem tend to resolve around software instead of hardware, would anyone have any leads on why something like this might be occurring? My best understanding of the subject is that the CPU core isn't being put into the correct state upon the RTC timer firing, thus leading to higher current draw. I'm ok with digging around more, but it would be useful to know if I'm the only one with the issue here. A quick search on the current project Issues and Pull Requests didn't yield anything useful.

**Should anyone be interested in the project source:** All my relevant code is in **sentry/src/**
[project_code.zip](https://github.com/SuperHouse/esp-open-rtos/files/2002364/project_code.zip)

projectgus, 2018-05-15:
Hi @brandonbraun653 ,

I don't know the precise answer to your problem, but in case it helps - I have noticed in the past that breadboards can cause surprisingly high amounts of current leakage just through their structure.

To fully rule out hardware vs software you could try using a framework like Arduino (built on a different Espressif SDK version) to go into deep sleep and compare the current draw for this.

brandonbraun653, 2018-05-15:
Hey! It's definitely worth a shot. We are at such low currents anyways that it certainly is in the realm of generic leakage. I have a custom board I've been using for my development system, but it's very hard to pick out the exact ESP chip current from the lump sum meter measurement. I'm about to spin off a new board design tomorrow and think I might add some precision low value resistors in series with all my active components power rails so I can get some decent measurements.

In addition, I've been playing with the power supply voltage and that can dramatically help. While only tested on the breadboard version, dropping from 3.3V down to 2.8V cut my deep sleep current down to 150 uA, so that's nice. My custom board worked fully at that voltage too.

brandonbraun653, 2018-06-04:
UPDATE: After redoing my board design and switching from a 12E to a 12F (which _shouldn't_ make a difference), I was able to get my current consumption down to about 70uA when powering everything off of 3.0V. Granted this was powering all my circuitry and not just the ESP chip, so I feel it is safe to say the problem somehow lies with the specific chip I got and not anything to do with esp-open-rtos.

closed 2018-06-04 by brandonbraun653

AndreaGreco, 2018-07-27:
Could you post source code. I want read it for checkout what you has showdown.
Your source could be a start point for set up example that show how setup ESP for low power project.