SuperHouse/esp-open-rtos #628: stop watchdog

Zaltora, 2018-05-28:
How can i stop watchdog or feed him in a infinite loop ?

I want shutdown my device properly. In a function, i suspend FreeRTOS, set some level pin and i want finish by a infinite loop until the device is off.

ourairquality, 2018-05-29:
Perhaps the function `sdk_wdt_init()` in `app_main.c` would give some clues. There is a hardware wdt and one of the mac timers is used for a soft wdt. Perhaps the following keep both fed:

```
WDT.FEED = WDT_FEED_MAGIC;
sdk_wDev_MacTim1Arm(1600000);
```

Zaltora, 2018-05-29:
it did not work. I added these two lines in the loop and add the include.
FreeRTOS must do it with the idle task or the tick hook.
I did not find or exactly FreeRTOS manages the watchdog.

ourairquality, 2018-05-29:
FreeRTOS does not appear to have anything to do with the wdt. The wdt appears to be fed from the wifi binary, the soft wdt from the mac counter. Perhaps the wifi layer is failing and hitting some assertion, e.g. if the NMI is disabled for too long it asserts. Perhaps try some of the following:

```
WDT.CTRL &= ~WDT_CTRL_ENABLE;
DPORT.INT_ENABLE &= ~DPORT_INT_ENABLE_WDT;
pp_soft_wdt_count = 0;
```

Zaltora, 2018-05-29:
i try this without succes:
```c
        while(1)
        {
            WDT.FEED = WDT_FEED_MAGIC;
            sdk_wDev_MacTim1Arm(1600000);
            sdk_pp_soft_wdt_feed();

            WDT.CTRL &= ~WDT_CTRL_ENABLE;
            DPORT.INT_ENABLE &= ~DPORT_INT_ENABLE_WDT;
            //pp_soft_wdt_count = 0;
        }
```
pp_soft_wdt_count seem unknow ?

Is not FreeRTOS taking care of the watchdog?
How can the wifi binary take the hand to reset the watchdog? 
I assumed that FreeRTOS was doing it because when I did a task without delay, it reboot.

Zaltora, 2018-05-29:
PS2: the code above make the system break;

```c
        while(1)
        {
           // WDT.FEED = WDT_FEED_MAGIC;
           // sdk_wDev_MacTim1Arm(1600000);
           // sdk_pp_soft_wdt_feed();

            WDT.CTRL &= ~WDT_CTRL_ENABLE;
            DPORT.INT_ENABLE &= ~DPORT_INT_ENABLE_WDT;
            //pp_soft_wdt_count = 0;
        }
```
don't work too. Automatic reset after some seconds

Zaltora, 2018-06-07:
i will try to enter in deep sleep mode to work around the problem.
I hope the gpio16 pin does not change state during the transition. (state:  0 logic)

SmartArduino, 2020-05-23:
Did you find a way to stop

SmartArduino, 2020-06-02:
This issue has been resolved because flash SPI speed