SuperHouse/esp-open-rtos #632: initialization problem during boot

paoloach, 2018-06-04:
Hi, 
I tried  to run the example examples/sysparam_editor but as soon as it start I got the error:

> [NOTE: No current sysparam region (initialization problem during boot?)]

and then a stack trace 

> ==>
> @R@0L,:+2QQ3*|ppPP@2A!|Pl
> Setting '@R@0L,' to binary data...
> rtos assert /home/paolo/esp2866/esp-open-rtos/FreeRTOS/Source/queue.c 1406
> abort() invoked at 0x40204743.

> Stack: SP=0x3fff4270
> 0x3fff4270: 0000000d 3fff43b8 00000000 40204746
> 0x3fff4280: 3fff3a60 00000000 3fff02d8 40204f68
> 0x3fff4290: 00000000 00000000 3fff0378 4020489e

this is the command I used:

`make` -j4 -C examples/sysparam_editor flash  ESPPORT=/dev/ttyUSB0 FLASH_SIZE=4

and the esptool flash_id  results is

> Manufacturer: e0
> Device: 4016
> Detected flash size: 4MB

Note: I have the same problem of system param region not initialized with the example/wificfg

In the user_init of my application I replicated the code into app_main.c:sdk_user_start function that I see it is  used to initialize the sysparam region and it seems to work fine but I don't understand why it doesn't work in the sdk_user_start function.

Regards,
Paolo

ourairquality, 2018-06-05:
It should not be necessary to initialize the sysparam in user_start. The problem is that the region did not initialize correctly, so it did not create the semaphore (we can fix that). I suggest tracing through `sysparam_init()` to see where it is failing and that might give some clues.

paoloach, 2018-06-05:
Hi, thank for the response, but now magically all works fine.
I had begun to use some static variable in order to understand the real execution flow and I noticed the problem was gone.  It reverted all my modification and again the problem was gone.
The only modification from yesterday is I disconected the esp2866 board, and so I powered off it, instead of simply reset it.

kanflo, 2018-06-30:
Feel free to reopen the issue if the problem comes back.

closed 2018-06-30 by kanflo

MatzeS, 2019-08-25:
If somebody experiences the same problems:
For me, running `esptool erase_flash` and reflashing the firmware resolved the issue.