SuperHouse/esp-open-rtos #650: wificfg example: Client doesn't get an IP address

MarcusLindemann, 2018-06-25:
I compiled the wificfg example and flashed it onto a Wemos D1 (in fact on several boards of those). Tried to connect to the wifihot spot, but the client never receives (or finishes reception) of an IP address. Tried with an Android phone, an Android  tablet and a an old Blackberry playbook. None succeeds within my time of patience (about a minute). Is there anything to be configured prior to building? 
I had poked around a little when I tried to integrate the wificfg library into a project of my own and it seems as if the dhcpserver never gets scheduled...

ourairquality, 2018-06-25:
Quickly tried my own fork on a Wemos D1 and the wificfg example did work, using the two Android devices I tried. Could you please give that a try for a start see https://github.com/ourairquality/esp-open-rtos Could try erasing the D1 flash completely, and perhaps even checking out the source again just to be sure. I have seen somewhat flaky AP connections at times, but not sure of the issues, but it usually works with a few retries.

MarcusLindemann, 2018-06-25:
Hi,
thanks for your quick reply. Completely erasing the flash memory and reflashing the SW did the trick. Apologies for not thinking about flash erasure myself.

closed 2018-06-25 by MarcusLindemann