SuperHouse/esp-open-rtos #656: Cannot do a software reset

morcillo, 2018-07-06:
Hello, I am doing a project in which I need to do a software reset. 

I have trird the following functions: sdk_system_restart, sdk_system_restart_enchance and sdk_system_restart_in_nmi (I tried doing the same thing as in the function post_crash_reset), but the board apparently hangs on reset. 

At the moment I am running the following code:
```

        printf("Restarting...\n");
        uart_flush_txfifo(0);
        uart_flush_txfifo(1);
        sdk_system_restart();
```

and I get the following response:

> Restarting...
> rm match
> del if0
> usl
> sul 0 0
> 
> ets Jan  8 2013,rst cause:2, boot mode:(1,6)
> 
> 
> ets Jan  8 2013,rst cause:4, boot mode:(1,6)
> 
> wdt reset

I am also using libesphttpd and I'm using the captive portal, but the thing is that not even the captive portal code is able to reset the board, even though it connects it needs a manual reset for it to start working again. Basically what I'm doing is I'm monitoring the status of the STATION connection, if it's not connected it will set the mode to STATIONAP and restart, that way the captive portal will be initialized and wait for a new connection. Basically, whenever I lose connection to my router it will reset and open the captive portal.

PS: I'm using a board I developed and a wemos D1 and neither is restarting correctly

phkehl, 2018-07-06:
sdk_system_restart() works for me. However, I have seen it not working right after flashing and the first startup. After manual reset and restarting via the reset button or power cycle it would work. Not sure why that is.

ourairquality, 2018-07-06:
With the default esptool it does not restart after a flash for me either, but does after a reset. When using a more recent esptool version this does not appear to be an issue.

morcillo, 2018-07-06:
Well .. @phkehl and @ourairquality 

Apparently it's the esptool as you said. I powercycled the board and it started working properly. Thanks a lot for your help

morcillo, 2018-07-06:
Issue resolved. 

- When using an older version of esptool it needs either a reset or a power cycle before it starts working after flashing. 

- Or simply use the most recent version of esptool

closed 2018-07-06 by morcillo