SuperHouse/esp-open-rtos #657: Flash address

hanyuzzw, 2018-07-18:
Hello, I want to konw if I use the `FLASH_DOWNLOAD_TOOLS` to flash the binary, where can I find the 0x00000.bin and what the address should I flash the program bin into? Because I tried different addr but all failed.

paoloach, 2018-07-21:
Hi, I will try to help you. 
Honestly I don't know what you mean for ' FLASH_DOWNLOAD_TOOLS': a specific program or a generic program to flash the chip so can be I totally don't understand you question.
Anyway, using esptool I have to put:
 at address 0x000 the bootloader that you can find at bootloader/firmware_prebuilt/rboot.bin.
at address 0x1000 a blanck config that you can find at bootloader/firmware_prebuilt/blank_config.bin
at address 0x2000 your program.

I hope I been useful,
Paolo

hanyuzzw, 2018-07-21:
`Flash download tool`  is a binary download tool that can be found from the official website，and I have got the answer,thank you very much for your reply!

I will close it.

closed 2018-07-21 by hanyuzzw