SuperHouse/esp-open-rtos #658: esp 8265  M2 wifi connection issues

mehdilauters, 2018-07-19:
Hi,
I am getting in troubles trying to use an [esp M2](https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8285---ESP-M2) mounted on a led controller [module](https://fr.aliexpress.com/item/Magic-Home-Mini-RGB-RGBW-Wifi-Controller-For-Led-Strip-Panel-light-Timing-Function-16million-colors/32686853650.html) . I have no issue when using the [tasmota](https://github.com/arendst/Sonoff-Tasmota) firmware based on arduino and platformio but I can't connect to any access point with ````esp-open-rtos````
I use the ````http_get```` example to test it.
Everything works well on a ESP12E module I use as reference.
On the ESP M2

- if I do not change anything on the makefile:
infinity loop of
````
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28): 
````
- if I set ````FLASH_MODE ?= dout```` on the makefile,
the firmware run without any error but never connect to my AP (infinity loop of dns lookup fails)
````
p_task_hdl : 3ffef808, prio:14, stack:512
pm_task_hdl : 3ffef270, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3818, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jul 19 2018 13:26:29
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(dc:4f:22:81:4f:a7)
add if0
HTTP get task starting...
Running DNS lookup for ipv6.google.com...
DNS lookup failed err=202 res=0
Running DNS lookup for ipv6.google.com...
DNS lookup failed err=202 res=0
````

Do you have any idea of what could be wrong?
Regards

paoloach, 2018-07-21:
Hi,
It seems to me that
1) you have to use dout flash mode to program (it depends on the boards (https://github.com/espressif/esptool/wiki/SPI-Flash-Modes)
2) Can be that the problem is ipv6.google.com and not your board. In the http_get example try to change the web server from ipv6.google.com to www.google.com.

Regards,
Paolo

mehdilauters, 2018-07-24:
Thank you for your help, but actually, the wifi station doesn't connect to the access point so it is not an ipv6 issue and the dhcp server does not display any traces

paoloach, 2018-07-25:
Another test you can do is using the wificfg example. It allows you to create a new wifi network (default mode) or to connect to a existing one.
See in the code for user / password.

mehdilauters, 2018-07-25:
Yes, but the same code works on another target (esp12e) so it should be a low level issue or something related to the hardware

mehdilauters, 2018-10-10:
Hi all,
Does anyone have any idea of what could be wrong,
As previously, a given code is working well on esp-12. The only change made on it is the DOUT flash mode to flash the esp-m2.
It correctly boots but:
- it still does not want to connect to any wifi essid (no scandone log)
- any [sysparam_*](https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/sysparam.h#L364) function fails

Regards