SuperHouse/esp-open-rtos #660: Reset on connection closing in tcp_non_blocking example

ayushchand, 2018-07-23:
I am getting reset while closing a socket connection. I am using the tcp_non_blocking example. The reset cause is rst cause:2, boot mode:(3,6), triggered from FreeRTOS/Source/queue.c . I am able to connect and send the data. How do I decode the exception in esp-open-rtos? Did anyone else face this issue? 

Pasting my serial output for reference

pm_task_hdl : 3ffef2f0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3818, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jul 23 2018 15:40:51
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
DHCP started
mode : softAP(f2:b4:29:c5:e9:20)
add if1
bcn 100
Server netconn 1073697880 ready on port 50.
Server netconn 1073698280 ready on port 100.
add 1
aid 1
station: 38:a4:ed:5f:a3:3f join, AID = 1
Client incoming on server 1073697880.
New client is 1073684624.
Client 1073684624 send: Data
Client 1073684624 send: Data
Error read netconn 1073684624, close it
rtos assert /home/ayush/CodeWorkspace/Esp-open-rtos/esp-open-rtos/FreeRTOS/Source/queue.c 1959
abort() invoked at 0x402048ff.

Stack: SP=0x3fff4ab0
0x3fff4ab0: 3fff39a8 3fff2090 00000000 40204902
0x3fff4ac0: 3fff39a8 00000001 3ffef130 402053a0
0x3fff4ad0: 3fff2090 3fff4b00 3fff20a0 40205479
0x3fff4ae0: 00000001 3fff1604 3fff2090 40216c71
0x3fff4af0: 3fff2090 3fff2090 00000000 40215cf6
0x3fff4b00: 3fff2090 3fff4b00 3fff2090 00006a4a
0x3fff4b10: 00000000 3fff4b84 00000001 00000000
0x3fff4b20: 00000000 00000002 3ffef240 40215f9c

Free Heap: 46104
_heap_start 0x3fff1698 brk 0x3fff5774 supervisor sp 0x40000000 sp-brk 43148 bytes
arena (total_size) 16604 fordblks (free_size) 2956 uordblocks (used_size) 13648

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffefa00, prio:14, stack:512
pm_task_hdl : 3ffef2f0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3818, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jul 23 2018 15:40:51
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
DHCP started
mode : softAP(f2:b4:29:c5:e9:20)
add if1
bcn 100
Server netconn 1073697880 ready on port 50.

ourairquality, 2018-07-24:
It might be a poor example. There is a double free occurring due to double calls to netconn_delete. Could you try using the socket API rather than the netconn API, otherwise you might have to explore how it should interact with threading - what if there are multiple queued events pointing to the same connection and one deleted that connection, and how to spot the rest using that connection etc.

ayushchand, 2018-07-26:
I have tried commenting netconn_delete, no use. I have verified the example
and couldn't find any fault in the api calls, I like to use netconn as they
have better documentation.

ourairquality, 2018-07-27:
There was a double free in src/api/api_msg.c `netconn_free()`, the call to  `sys_sem_free()` and it is probably failing in the same way, so that is a fault. You might need to work out how that occurs to understand the api issue.

I see that in the example `netCallback()` stores a pointer to a `struct netconn` in the events queue. Might that be a problem if there were multiple events queued and one deletes the netconn and then the next even tries to re-use that deleted netconn. There might be other issues.