SuperHouse/esp-open-rtos #666: Old SD card doesn't support CMD25, WRITE_MULTIPLE_BLOCK

chen-hsiang0020, 2018-08-02:
I noticed that sdio_write_sectors() return error when I using my lower end SD cards. My no-brand 128Mb and 256Mb cards having issue on f_mkdir() and the root cause is error in sdio_write_sectors(). To fix the issue, I have to replace the CMD25 (WRITE_MULTIPLE_BLOCK) with multiple CMD24 (single WRITE_BLOCK) to overcome the issue.
To increase the compatability on different SD Cards, may I propose the change to replace CMD25?

(sdio_write_sectors() in extras/sdio/sdio.c)
```
#if 0
    if (command(card, CMD25, sector))
        return set_error(card, SDIO_ERR_IO);

    while (count--)
    {
        if (write_data_block(card, TOKEN_MULTI_TRAN, src) != SDIO_ERR_NONE)
            return card->error;
        src += SDIO_BLOCK_SIZE;
    }
    spi_transfer_8(BUS, TOKEN_STOP_TRAN);
#else
    // CHEN: Lower end card (EX: 128Mb) doesn't support
    //       multiple block write (CMD25)
    while (count--)
    {
        // single block
        if (command(card, CMD24, sector)) {
            return set_error(card, SDIO_ERR_IO);
        }
        if (write_data_block(card, TOKEN_SINGLE_TRAN, src) != SDIO_ERR_NONE)
            return card->error;
        src += SDIO_BLOCK_SIZE;
    }
#endif

```

UncleRus, 2018-08-02:
Workarounded in #667, please test

UncleRus, 2018-10-17:
Feel free to reopen issue if the error persists

commit 253ac079 (2018-10-17): Merge pull request #667 from UncleRus/sdio_cmd25_fix

CMD25 workaround for SDIO, fix #666
closed 2018-10-17 by UncleRus