SuperHouse/esp-open-rtos #668: Rom not able to boot at different position than 0x2000

apiel, 2018-08-03:
On one of my esp8266 (sonoff) the Rom is only able to boot at position 0x2000. So even If I upload a new rom with OTA at the second address (romconf->roms[1]) the mcu will keep booting at 0x2000 even if the setting of romconf->current_rom are set to 1.
To solve this issue I updated rboot to copy the rom at the address romconf->roms[1] to 0x2000. Then it boot to the new rom. Is there any reason this happen? How to solve this without to customize rboot?

This is a part of what I did in rBoot (still not very clean for the moment but it s working):
```
static uint8 move_rom_to_pos_0(uint32 readpos) {
	uint32 readposCopy = readpos;
	uint8 buffer[BUFFER_SIZE];
	uint8 sectcount;
	uint32 romaddr = get_rom_addr(&readpos, &sectcount);

	uint32 romSize = sectcount * SECTOR_SIZE;
	uint32 newRomAddr = (BOOT_CONFIG_SECTOR + 1) * SECTOR_SIZE;
	// uint32 end = romSize + newRomAddr;
	uint32 end = readposCopy + 300000;
	ets_printf("Loop at %d copy of %d till %d\n", newRomAddr, readposCopy, end);
	for (; readposCopy < end; newRomAddr += BUFFER_SIZE, readposCopy += BUFFER_SIZE) {
		// ets_printf("Yoyo\n");
		ets_printf("Write at %d from %d\n", newRomAddr, readposCopy);
		if (SPIRead(readposCopy, buffer, BUFFER_SIZE) != 0) {
			ets_printf("read err %d\n", readposCopy);
			return 2;
		}
		ets_printf("Erase %d %d\n", newRomAddr % SECTOR_SIZE == 0, newRomAddr);
		if (newRomAddr % SECTOR_SIZE == 0 && SPIEraseSector(newRomAddr / SECTOR_SIZE) != 0) {
			ets_printf("erase err\n");
			return 3;
		}
		ets_printf("Write\n");
		SPIWrite(newRomAddr, buffer, BUFFER_SIZE);
	}
	ets_delay_us(999999);
	return 0;
}
```
Unless I find a better solution to make the rom boot at a different address with rboot, I guess I will write my own bootloader. But of course it would be better to keep rboot as it is ;-)

AndreaGreco, 2018-08-04:
You could checkout `extras/http_client_ota/http_client_ota.c` as sample. This is a ota HTTP client.
This flash firmware and setup bootloader rom area.
use `rboot_set_current_rom(rom_to_use) for setup next bot rom.
with `rboot_config = rboot_get_config();`
rboot_config give you other useful info as where is rom page address...
Check rboot_config struct and how other firmware use it.

apiel, 2018-08-04:
Hi, thx for your help. I already tried several time this example, I was actually using it to implement OTA in my firmware. It's actually interesting, cause this example doesn't seem to have issue with sdk_spi_flash_erase_sector where this was one of my problem. Anyhow, the example is also failing and not able to reboot. But with the bootloader I wrote yesterday, it work: https://github.com/apiel/aboot

So I guess at some point there might be some memory issue in esp-open-rtos and of course, I might also make some wrong stuff in my firmware. I wish I could find the reason of all this problem and provide some improvement ;-)

HomeACcessoryKid, 2019-03-09:
For another example using temporary boot you can look at https://github.com/HomeACcessoryKid/life-cycle-manager. This saves the sector 0x1000 from being written to.