SuperHouse/esp-open-rtos #670: BMP280 Reading values

TrigerxD, 2018-08-05:
Hi. I've wrote some code to get measures from bme280 sensor but one moment it only display values : 
Temperature : 21.74 (when in my room is ~28)
Pressure : 68813.25 (something terrible...)
Humidity : 67.47 (quite normal)
But that values appears always...
Anybody know how to fix it?

Zaltora, 2018-08-06:
I got maybe the same problem as you. I got 2 sensors (bme280).
First:
T1:22.31
H1:69.65
P1:63095.65     
Second:
T2: 23.26
H2:70.61
P2:63358.26

My settings are for both:
```c
    static bmp280_params_t _xParams;  //Store bme280 parameters
    _xParams.mode = BMP280_MODE_NORMAL;
    _xParams.filter = BMP280_FILTER_2;
    _xParams.oversampling_pressure = BMP280_STANDARD;
    _xParams.oversampling_temperature = BMP280_STANDARD;
    _xParams.oversampling_humidity = BMP280_STANDARD;
    _xParams.standby = BMP280_STANDBY_1000;
```
what are yours ?
my first guess is : sensor is dead (because he was working before, and no HW change)
my second guess : i2c lib  (test with different freq / commit. still not work)
The max frequency with 80MHz system clock is ~320KHz, even if you set more. I don't remenber the minimum frequency with bme280.
the last guess: the library somewhere... a miss calculation, wrong configurations...

I just hope that some wrong i2c setting don't destroy the component by writing in some registers...

TrigerxD, 2018-08-06:
I found my problem thanks to you. I have set the i2c clock to 400k what was a problem with incorrect readings.  I've set it back to 100k and now it measure ok :) Look in i2c init parameters and try changing something in it :)

Zaltora, 2018-08-06:
nope, nothing change for me.
Strange solution... BME280 I2C frequency max is 3.4MHz so 400KHz is fine.
Are you sure that you only change the i2C frequency ? your bme280 setting is different ?

Zaltora, 2018-08-06:
Found the problem. pointer on param fail... Now it is work and it is work at 400KHz.