SuperHouse/esp-open-rtos #672: ESP8266_RTOS_SDK uses Apache 2.0 license

tkremeyer, 2018-08-08:
As of this april, ESP8266_RTOS_SDK seems to use an Apache 2.0 license (see [this commit](https://github.com/espressif/ESP8266_RTOS_SDK/commit/19aed2e67ffcc27ba4dca9aad4b4264f62c4bcc2)).  

I am not that much into the dependencies and licensing of esp-open-rtos, but I wonder if that means that we can use current versions of Espressif SDK again (or is it a problem that [ESP8266_NONOS_SDK](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/License) still uses Espressif MIT?). Therefore, I would be grateful if a maintainer could look into this. It would be amazing to be able to use the optimizations and new features introduced with the new SDK versions at some point in the future - while I'm aware that this update would be a huge piece of work.

flannelhead, 2018-08-08:
What's equally interesting, the SDK now seems to use [lwIP 2.0.3](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/lwip/lwip/CHANGELOG) and [FreeRTOS 10.0.1](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/freertos/include/freertos/FreeRTOS.h#L2). I wonder if it could help with adopting the binaries.

ourairquality, 2018-08-08:
Fwiw  I made a small start exploring this, moved some esp-open-rtos extras into 'components', some sensors drivers and the wificfg, and also built a current newlib to replace the version in ESP8266_NONOS_SDK. See numerous issues that will take time to work through, but it seems a promising direction.

chen-hsiang0020, 2018-08-09:
That will be wonderful to use the latest binary from Espressif SDK. For my opinion, updating the binary is the key to have more serious projects to select esp-open-rtos. Knowing that bugs are fixed in the new SDK but nothing we can do here in esp-open-rtos is frustrating, and missing security fixes will be the blocking issue for any commercial use of esp-open-rtos.

Zaltora, 2018-08-11:
It is nice. I had plan to replace the API calls (SPI / I2C / PWM ..) by an API defined in a file that is configure for the current SDK to make the library portable. This is a good opportunity to start working on it.