SuperHouse/esp-open-rtos #674: dangerous relocation: call0: call target out of range:

andrey13baciu, 2018-08-15:
Hi!
I have an issue with the linker when I have a big code grater then ~544k.
** -mtext-section-literals  -mlongcalls** options are used.

I always get the error:
_**"LD build/blink.out
./build/sdklib/libwpa.a(wpa_main.o): In function `sdk_eagle_auth_done':
(.irom0.text+0x295): dangerous relocation: call0: call target out of range: sdk_eagle_auth_done
collect2: error: ld returned 1 exit status"**_

The memory is configured to use 1MB as I can see from .map file:
**Name             Origin             Length             Attributes
dport0_0_seg     0x000000003ff00000 0x0000000000000010
dram0_0_seg      0x000000003ffe8000 0x0000000000014000
iram1_0_seg      0x0000000040100000 0x0000000000008000
irom0_0_seg      0x0000000040202010 0x00000000000fdff0
       default        0x0000000000000000 0xffffffffffffffff**

To replicate the error I used 4 const strings of 100000 bytes then I print them after UART_init in Blink example. The same error is returned in a bigger project which doesn't have such ridiculous buffers.
If I print only 3 strings everything works.

I searched a lot but I couldn't find any solution to my problem.
Any help would be greatly appreciated!
    Thank you!

JDRobotter, 2018-08-21:
Same problem here, on a large code base.

andrey13baciu, 2018-08-22:
I tried official SDK from Espressif, I used the same method where I changed an example and used some big constants and it seems to work for big code.
I still hope that there is a way to make SuperHouse work on the same way.
I tried a lot of things, but it seems that it is beyond my knowledge.

ourairquality, 2018-08-22:
Could you try renaming that symbol, so change lib/allsymbols.rename
eagle_auth_done sdk_eagle_auth_done_x

We have source for this now in open_esplibs/libwpa/wpa_main.c Not sure why the linker has this issue, but renaming that symbol has worked around this for me.

andrey13baciu, 2018-08-22:
I changed the symbol and it seems that now I have more space. Now I can successfully build 640k, but if i increase the size to 723k I receive more errors of this kind

./build/sdklib/libnet80211.a(ieee80211_hostap.o): In function `sdk_hostap_input':
(.irom0.text+0xede): dangerous relocation: call0: call target out of range: sdk_ieee80211_hostap_attach
./build/sdklib/libmain.a(user_interface.o):(.irom0.text+0x15e): dangerous relocation: call0: call target out of range: sdk_system_restart.... 
(I attached all the output)

Should I rename all the symbols with "dangerous rellocation: call0: call target out of range" errors?

[Build debug.txt](https://github.com/SuperHouse/esp-open-rtos/files/2310463/Build.debug.txt)

andrey13baciu, 2018-08-29:
I tried a lots of things, but I couldn't solve the problem. Thanks to @ourairquality  I can now build up to 640k instead of 544k, but nothing more.
I tried even to rename the symbols that  return dangerous allocation, but it seems that this has no source code.
Is there a way to solve this problem for the precompiled libraries?

JDRobotter, 2018-08-29:
It seems the binary size is not strictly related to the linker failure, because my binary is currently ~673k total and I did not follow @ourairquality advice.
```
   text	   data	    bss	    dec	    hex	
 634570	   2504	  36664	 673738	  a47ca	
```
Yet for what I see I am very close to the edge and I dont think I can fit more than a few 10k before some symbols fails to be relocated.

pavels, 2018-11-18:
Probably found it
It happens, when the code is large and some stuff from sdklib ends too far in flash - the main problem is that the binary blobs used is built without `-mlongcalls`

Don't know if it is 100% percent solution but works for me - you need to force sdklib stuff at the beginning of flash - this can be done with following linker script patch

```diff
--- a/ld/program.ld
+++ b/ld/program.ld
@@ -211,6 +211,7 @@ SECTIONS
      *****************************/
 
     _irom0_text_start = ABSOLUTE(.);
+    *sdklib*:*(.irom.* .irom.*.* .irom0.*)
     /* esp-open-rtos compiled code goes into IROM by default
        (except for libgcc which is matched above.)
 ```

andrey13baciu, 2018-11-19:
Hi!
@pavels solution worked for me too!
Thank you!

closed 2018-11-20 by andrey13baciu

kiralikbeyin, 2019-06-06:
I added to parameters.mk 
````
CXXFLAGS	+= -mlongcalls
CFLAGS += -mlongcalls
C_CXX_FLAGS += -mlongcalls
````

but still getting 
````
./build/sdklib/libnet80211.a(wl_cnx.o): In function `sdk_cnx_add_rc':
(.irom0.text+0xc6a): dangerous relocation: call0: call target out of range: sdk_cnx_remove_rc
./build/sdklib/libnet80211.a(wl_cnx.o): In function `sdk_cnx_update_bss':
(.irom0.text+0xf30): dangerous relocation: call0: call target out of range: sdk_cnx_remove_rc
./build/sdklib/libnet80211.a(wl_cnx.o): In function `sdk_cnx_sta_leave':
(.irom0.text+0x11db): dangerous relocation: call0: call target out of range: sdk_cnx_node_remove
./build/sdklib/libwpa.a(wpa_main.o): In function `sdk_eagle_auth_done':
(.irom0.text+0x295): dangerous relocation: call0: call target out of range: sdk_eagle_auth_done
collect2: error: ld returned 1 exit status
`````

where is right place to put -mlongcalls?

pavels, 2019-06-06:
@kiralikbeyin won't help, the problem can only be solved by changing linker script (`esp-open-rtos/ld/program.ld`) - https://github.com/SuperHouse/esp-open-rtos/issues/674#issuecomment-439717688

Long answer - you would need to rebuild sdklib with `-mlongcalls` but you can't do it (neither me), because we don't have source code for this part, it is binary blob supplied by espressif

kiralikbeyin, 2019-06-06:
Sadly it is not working for me...

[image]

pavels, 2019-06-06:
@kiralikbeyin hmm than i don't know, but it definitely solved this error for me

also the indentation in your example is sligthtly different , but i have no idea if that can be the problem

kiralikbeyin, 2019-06-06:
@pavels thanks, indentation is same as yours now but nothing..
Do you have any idea about HTTPS PUT https://github.com/SuperHouse/esp-open-rtos/issues/721

ourairquality, 2019-06-06:
We have source code for these. e.g. `sdk_eagle_auth_done` is defined in open_esplibs/libwpa/wpa_main.c   Perhaps it is a caller site in the binary sdk libraries that causes trouble. e.g. there is a call to `sdk_eagle_auth_done` in `sdk_ieee80211_sta_new_state` but even if we add source code for that then `sdk_ieee80211_sta_new_state` might start giving trouble. It might be possible to organize the linking order to place these trouble function closer to their callers?