SuperHouse/esp-open-rtos #675: taskENTER_CRITICAL() question

Zaltora, 2018-08-20:
Simple question, 
Use `taskENTER_CRITICAL()` and `taskEXIT_CRITICAL()` disable interrupt ?
It didn't disable interrupt from GPIO or TIMER ?

AndreaGreco, 2018-08-24:
It should disable all interrupt.
You should used just for few operation, that must be atomic. It disable interrupt execution.
Interrupt is not lost, it will be executed immediately after call exit critical.

Zaltora, 2018-08-25:
okay, thank you ! I use it for 2 operation to get time and a measure in an atomic way. I was afraid that interrupt was lost. A second solution for me was this : 
```c
static bool flag_atomic;

void vApplicationTickHook(void)
{
  flag_atomic = false;
}

void myFunction(void)
{
   do {
         flag_atomic = true;
         // Operation 1 here
         // Operation 2 here
         // ....
   while(!flag_atomic);
}

```

closed 2018-08-25 by Zaltora