SuperHouse/esp-open-rtos #680: AWS IOT Example does not work

BenediktMiller, 2018-09-14:
Hey, 

I configured and uploaded the aws_iot example. Unfortunately, the program only sends the following via the serial port:

```
mqtt_task: started
mqtt_task: connecting to MQTT server <myiotbroker>.iot.eu-central-1.amazonaws.com ... Error: -15744
error: -15744
````

I can't find anything about the error -15744 on the internet, is there a list of error codes available.

Any help would be greatly appreciated.

Best regards,
Benedikt

BenediktMiller, 2018-09-14:
Okay there was a problem with my private key file that I solved, now I get the following errors:

```
Schedule to publish
mqtt_task: started
mqtt_task: connecting to MQTT server <mybrooker>.iot.eu-central-1.amazonaws.com ... Error: -30592
Error: -29312
Schedule to publish
Error: -76
```

I think they are all error codes from mbedtls, but how can I get the error messages for the codes, how can I print the codes in hex?

I already found this file:https://github.com/ARMmbed/mbedtls/blob/development/include/mbedtls/error.h#L28

But it didn't help me in finding the error messages

uroshekic, 2020-05-11:
Removing the "-ats" part actually worked for me after. Is there an explanation for this?

AR-Tawil, 2020-05-11:
Hi, I'm still struggling to make the aws_iot example to work :) 
Are you using RSA for your client key? 
I'm getting a -9984 error?

uroshekic, 2020-05-11:
I was getting -9984 as well. :) No, RSA key & cert didn't work. I had to generate ECC-based certificate as described in the README file of the example. After this, I tested it with:
`mosquitto_pub --cafile AmazonRootCA1.pem  --cert eccCert.crt --key ecckey.key -h ...-ats.iot.eu-central-1.amazonaws.com -d -p 8883 -q 1 -t "foo/bar" -i test -m “HelloWorld”`
and it worked. 
I also had to manually link add the certificate to my device via AWS IoT Console.

On [AWS](https://aws.amazon.com/blogs/iot/elliptic-curve-cryptography-and-forward-secrecy-support-in-aws-iot-3/), there are instructions how to test IoT with mosquitto_pub and ECC-based certificates, and it says to set the following parameter `--tls-version tlsv1.2`, but this did not work in my case (TLS error)!

But for the aws_iot example, I had to go to AWS IoT Console -> Security, and download the certificate again (which was different than my original eccCert.crt, but I used the same key). Then, after removing "-ats" from client_endpoint, it finally worked.

klaslofstedt, 2020-05-12:
This is likely not helpful, but I did definitely end up using RSA successfully. I can't remember if i had to configure anything else.

AR-Tawil, 2020-05-13:
Thank you very much @uroshekic for your reply. 
I have followed the steps exactly as described in your message but it's still not working. I'm able to send the msg with mosquitto_pub  but unable to send message programatically. I'm always getting
mqtt_task: connecting to MQTT server xxxxx.iot.eu-west-2.amazonaws.com ... Error: -9984
error: -9984
Schedule to publish

I have tired many times but it's not working :(

nulcypher, 2020-05-19:
So - I have had a go at getting this working myself over the last few days. I don't recognise your error code but here is what I did and in some cases reverted.

As a good AWS citizen I thought I would try the -ats (Amazon Trust Services) endpoints. I may try and do this again but don't do it yourself without instruction - you'll surely have to model with the trust anchor in the code. The CA cert embedded in the code currently is suitable for the regular endpoint non-ATS endpoint so use them for now.

I actually run my IoT stuff on a heavily firewall VLAN... superstitious. I had forgotten to open up the MQTT port.

The hardest to solve which took a whole day of Googling and head scratching relates to the overloaded use of the CN (Common Name) RDN of a DN in TLS. This DN is built from all the stuff you are prompted for by "openssl req ..." - I think maybe the TLS library is doing something to establish trust that the cert is issued to the FQDN that is using it. I fixed by using the openssl command line to avoid this 'wizard' and supply a DN with no CN thus: 

openssl req -new -sha256 -key ecckey.key -nodes -out eccCsrNew.csr -subj "/C=GB/ST=Greater London/L=London/O=Test/OU=Test"

All now works for me.

I see in your other issue thread that you tried RSA keys because of the AWS API supplying them. While a good idea and you may be able to get it working I would also stick to the steps described in the README.

Cheers - Steve