SuperHouse/esp-open-rtos #682: sdk_wifi_station_scan not working properly for 32 character SSID

ja2142, 2018-10-09:
I'm not sure if this is a bug or not, but sdk_wifi_station_scan doesn't seem to work properly for 32 character SSID's. Example output from wifi_scan example:

```
----------------------------------------------------------------------------------
                             Wi-Fi networks
----------------------------------------------------------------------------------
                    UPC240176871 (70:85:c6:e5:5b:7e) RSSI: -87, security: WPA2/PSK
7GPowerOverWifi_MSDos_compatible␆�␃ (38:43:7d:8e:3a:32) RSSI: -44, security: WPA2/PSK
  012345678901234567890123456789 (b0:89:00:22:00:48) RSSI: -42, security: WPA2/PSK
```
Second wifi name should be "7GPowerOverWifi_MSDos_compatible", but there are extra unprintable characters at the end. 

Connection to said wifi works fine, but the extra characters are printed on connection:

`connected with 7GPowerOverWifi_MSDos_compatible␃␅␃, channel 6`

One of the causes is probably in include/espressif/esp_sta.h:
```
struct sdk_station_config {
    uint8_t ssid[32];     /* Null terminated string */
    uint8_t password[64]; /* Null terminated string */
    uint8_t bssid_set;    /* One if bssid is used, otherwise zero. */
    uint8_t bssid[6];     /* The BSSID bytes */
};
```
All ssids are stored as uint8_t[32], which means that the 32 byte ssid shouldn't fit.

Another thing is the fact that in wifi_scan example any characters after 32 should be cut by:
`printf("%32s (" MACSTR ") RSSI: %02d, security: %s\n", ssid,
MAC2STR(bss->bssid), bss->rssi, auth_modes[bss->authmode]);`
This means that printf string limiting isn't working. Indeed 
`printf("%2s","0123456789");`
outputs
`0123456789`
However
`%.*s`
format seems to be working as expected.

virangjhaveri, 2018-12-23:
printf function prints until '\0' is found. Since the SSID is exact 32 and the char array to store is also 32, the print function goes beyond and prints from memory location after the end of the variable. Hence the garbage character is printed.

phkehl, 2018-12-23:
`"%32s"` will pad 32 characters (spaces), but it will still print more than 32 characters if the string is longer.
Copy the string into a temporary string and nul-terminate it yourself:
```
char str[33];
strncpy(str, ssid, 32);
str[32] = '\0';
printf("ssid=[%s]", str);
```

ja2142, 2018-12-23:
Ok, so I was mistaken with printf format for cutting.

This doesn't change the fact that  `sdk_wifi_station_scan` gives too long strings for 32 byte SSIDs. I know that I could cut the extra bit elsewhere, but I belive that I shouldn't have to do that, because that's what library is for.

That being said `sdk_wifi_station_scan` is provided as a binary and I have no idea how to modify it.

phkehl, 2018-12-23:
Yeah, well. There are weirder things than this one. When I stumbled over this my though was that the "null terminated string" comment in the struct sdk_station_confg is just wrong. It's an array of bytes, not a string.

Here's my full scan-and-print-all-access-points function:
```
void wifiScan(void)
{
    printf("wifi: start scan\n");
    esp_err_t err;
    esp_wifi_scan_stop();
    wifi_scan_config_t scanConfig = { .show_hidden = true, /*.scan_type = WIFI_SCAN_TYPE_PASSIVE*/ };
    err = esp_wifi_scan_start(&scanConfig, true);
    if ( err != ESP_OK )
    {
        printf("wifi: scan failed: %s\n", esp_err_to_name(err));
        return false;
    }

    uint16_t numAp;
    err = esp_wifi_scan_get_ap_num(&numAp);
    if (err != ESP_OK )
    {
        return false;
    }
    printf("wifi: done scan (%u ap) %u\n", numAp, sizeof(wifi_ap_record_t));
    if (numAp > 50)
    {
        print("wifi: too man access points\n");
        return false;
    }
    wifi_ap_record_t *pApRecs = malloc(numAp * sizeof(wifi_ap_record_t));
    if (pApRecs == NULL)
    {
        printf("wifi: malloc\n");
        return false;
    }
    err = esp_wifi_scan_get_ap_records(&numAp, pApRecs);
    if (err != ESP_OK)
    {
        free(pApRecs);
        return false;
    }
    for (int ix = 0; ix < numAp; ix++)
    {
        const wifi_ap_record_t *pkAp = &pApRecs[ix];
        char country[4]; // there can be weird shit in pkAp->country.cc
        country[0] = isprint((int)pkAp->country.cc[0]) != 0 ? pkAp->country.cc[0] : '?';
        country[1] = isprint((int)pkAp->country.cc[1]) != 0 ? pkAp->country.cc[1] : '?';
        country[2] = isprint((int)pkAp->country.cc[2]) != 0 ? pkAp->country.cc[2] : '?';
        country[3] = '\0';
        char ssid[33];
        strncpy(ssid, (const char *)pkAp->ssid, 32);
        ssid[32] = '\0';
        printf("ap %02d/%02d "MACSTR" ch %2u%c rssi %3d %-12s %-9s %-9s %c%c%c %c %s"/* %c%c%c */" %s\n",
            ix + 1, numAp,
            MAC2STR(pkAp->bssid),
            pkAp->primary, pkAp->second == WIFI_SECOND_CHAN_ABOVE ? '^' : (pkAp->second == WIFI_SECOND_CHAN_BELOW ? '_' : '-'),
            pkAp->rssi, wifiAuthModeStr(pkAp->authmode), wifiCipherTypeStr(pkAp->pairwise_cipher),
            wifiCipherTypeStr(pkAp->group_cipher),
            (pkAp->phy_11b ? 'B' : '.'), (pkAp->phy_11g ? 'G' : '.'), (pkAp->phy_11n ? 'N' : '.'),
            (pkAp->wps ? 'W' : '.'), country, /*pkAp->country.cc[0], pkAp->country.cc[1], pkAp->country.cc[2],*/
            (pkAp->ssid[0] == 0 ? "<hidden>" : ssid));
    }
    free(pApRecs);
}
```

ja2142, 2019-01-03:
So it seems that ssids in sdk are null terminated only when less then 32 bytes in length. This leaves the user with limiting ssid's length. That's fine, but then the usages are wrong - two that i know of are wifi_scan example and `cgiWifiScan` from libesphttpd/util/cgiwifi.c. Example is harmless, but libesphttpd case adds some characters and the characters added may be illegal for JSON and make it unparsable (which happens for me sometimes). To fix this issue I guess I should do PR to https://github.com/nochkin/libesphttpd, right? Or should I PR original libesphttpd https://github.com/Spritetm/libesphttpd