SuperHouse/esp-open-rtos #684: lwip compiling issues

pablogs9, 2018-10-18:
I had this compilation error while following the compiling example:

make: Entering directory '/home/pablogs/homekit8266/esp-open-rtos/examples/http_get'
CC /home/pablogs/homekit8266/esp-open-rtos/open_esplibs/libmain/user_interface.c
CC /home/pablogs/homekit8266/esp-open-rtos/open_esplibs/libmain/spi_flash.c
CC /home/pablogs/homekit8266/esp-open-rtos/open_esplibs/libmain/ets_timer.c
CC /home/pablogs/homekit8266/esp-open-rtos/open_esplibs/libmain/os_cpu_a.c
In file included from ../../core/include/sdk_internal.h:4:0,
                 from /home/pablogs/homekit8266/esp-open-rtos/open_esplibs/libmain/spi_flash.c:14:
../../include/espressif/esp_wifi.h:14:26: fatal error: lwip/ip_addr.h: No such file or directory
 #include <lwip/ip_addr.h>

blissboy, 2018-12-04:
I also get this error

blissboy, 2018-12-04:
https://github.com/SuperHouse/esp-open-rtos/wiki/Troubleshooting

closed 2020-03-03 by pablogs9