SuperHouse/esp-open-rtos #685: esp8266 STA mode can't assign nameq

mohdamerkhalidi, 2018-11-12:
Hi,
  I am not able to assign a hostname to the esp8266. I have made changes in the esp_interface.c file's ethernetif_init() function's sysparam_get_string("hostname", &hostname); I have made sure that LWIP_NETIF_HOSTNAME is also enabled. But still, if I give hostname.local in the browser, nothing happens. Whereas if I give the IP address, I am able to connect to the esp8266 from the browser. I am working on the http_server example of esp-open-rtos.
   Can someone please suggest what I might be missing here.

Thanks.

phkehl, 2018-11-12:
I was struggling with that, too. In STA mode, though. I've ended up with this:
`    // doesn't seem to work in wifiInit() (user_init())
    //sdk_wifi_station_set_hostname(sWifiData.staName);
#if LWIP_NETIF_HOSTNAME
    struct netif *netif = sdk_system_get_netif(STATION_IF);
    sdk_wifi_station_disconnect();
    netif_set_hostname(netif, sWifiData.staName);
    sdk_wifi_station_connect();
#endif
`
sWifiData.staName has my desired hostname.
See https://github.com/phkehl/tschenggins-laempli/blob/master/src/wifi.c#L643

mohdamerkhalidi, 2018-11-12:
Thank you for the reply. I'll try this today.

skcks, 2018-11-12:
@phkehl try this code, it work for me

```
    struct netif *netif = sdk_system_get_netif(STATION_IF);
    netif_set_hostname(netif, gp_config->device.dev_name);
    dhcp_release(netif);
    dhcp_start(netif);
```

mohdamerkhalidi, 2018-11-12:
Hi,
  Thanks for the suggestions. I tried both the ways but I am still not able to connect.. These are my steps in order - 
struct sdk_station_config config;
memcpy(config.ssid, ptr_ssid, 32);
memcpy(config.password, ptr_ssid_password, 64);
sdk_wifi_set_opmode(STATION_MODE);
sdk_wifi_station_set_auto_connect(TRUE);
sdk_wifi_station_set_config(&config);
while(1)//while loop until I get STATION_GOT_IP
{
         wifi_status = sdk_wifi_station_get_connect_status(); 
}
struct netif *netif = sdk_system_get_netif(STATION_IF);
sdk_wifi_station_disconnect();
netif_set_hostname(netif, "my_esp");
sdk_wifi_station_connect();
      Once I get the IP, from the browser I give Http://my_esp.local. I don't get any response at esp side. 
I have also tried what @skcks suggested but I am not getting anything at esp.

mohdamerkhalidi, 2018-11-13:
Hi @skcks,
   I will not be able to test it right now. But, can you help me with the steps. This is what I'll try to do - 

struct sdk_station_config config;
memcpy(config.ssid, ptr_ssid, 32);
memcpy(config.password, ptr_ssid_password, 64);
sdk_wifi_set_opmode(STATION_MODE);
sdk_wifi_station_set_auto_connect(TRUE);
sdk_wifi_station_set_config(&config);
while(1)//while loop until I get STATION_GOT_IP
{
wifi_status = sdk_wifi_station_get_connect_status();
}
mdns_init();
mdns_add_facility("my_esp", "_http", "", mdns_TCP+mdns_Browsable, 443, 600);
  Then try to connect using  Http://my_esp_http.local. Is this correct?

skcks, 2018-11-13:
@mohdamerkhalidi it is ok, if your http_server listen on 443. otherwise change 443 to the actual port http server listening.

espwork, 2018-11-27:
Mdns doesn't work with esp-open-rtos. Arduino's mdns works fine for the host name. Wish there was a simple example for the hostname.

espwork, 2018-11-28:
Can someone please provide a simple sample code for setting the hostname. I have tried with wificfg too and that too is not working.