SuperHouse/esp-open-rtos #699: Is this repo slowly dying?

SaimenSays, 2019-02-17:
Don't know where to put my question so I do it here...

There are many open pull requests, but the commit frequncy goes down every month.
I don't want to blame anyone for this. It is great work, what was done here! Thanks for all the effort of the maintainers. This is way the best platform for ESP8266.

My question is, if the ESP8266 is less used these days? Is everyone skipping Arduino? Or is the ESP32 taking place for new projects?
Can I count on esp-open-rtos for new projects, or can anyone tell me where the the developers are tending to?

skcks, 2019-02-18:
maybe the official sdk is better for new project.

SaimenSays, 2019-02-18:
@skcks: I tried original RTOS longer time ago, but it lacks in many features (e.g. RAM usage). Has this changed? Do you think this repo is going down because of the increased usability of original espressif one?

chen-hsiang0020, 2019-02-19:
I feel that it is because the esp-open-rtos is locked in to the way-too-old Espressif sdk binary. So many bugs in the GitHub are due to the old binary it based on, and most of the conclusions are “nothing we can do about it”. It making me nerves on knowing that critical bugs are unresolvable. Really hope that someone can take the lead to update the binary….

Fonger, 2019-09-07:
[ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK) is now under standard [Apache License 2.0](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/LICENSE). Hence, the reason why esp-open-rtos used old Espressif binary is gone.

We definitely can use its latest binary to add new features and solve bugs. However it requires a lot of effort to update this project. I really hope someone can do this one day. For my, my major concern is that Modem sleep is not support in esp-open-rtos and some serious security vulnerability like KRACK is not patched in esp-open-rtos.

Fonger, 2019-09-12:
There is one more critical vulnerabilities `CVE-2019-12588` found this year.
https://matheus-garbelini.github.io/home/post/esp8266-beacon-frame-crash/

Still, the latest SDK has fixed this but it is not possible to fix in esp-open-rtos.
It's insecure to use in production especially if your esp8266 is accessible in public.

Fonger, 2019-09-12:
@SaimenSays 

I'm trying to port every projects I works to latest ESP8266_RTOS_SDK 3.2 from esp-open-rtos.
What do you mean by RAM usage? Do you mean get the free available heap?
If so, yes, it has ram usage `esp_get_free_heap_size()` which returns available heap size bytes in latest SDK. (equivalent to `xPortGetFreeHeapSize()` in esp-open-rtos)

The "SDK difference section" in wiki of esp-open-rtos is outdated too. 
Now both esp-open-rtos and ESP8266_RTOS_SDK 3.2 functions stored in flash by default. And if you want to move it to IRAM, you can add prefix `IRAM_ATTR` (equivalent to `IRAM` in esp-open-rtos`)

SaimenSays, 2019-09-15:
I started with ESP8266 on original SDK years ago, then we realized that there was very view RAM left for user application. String constant were loaded to RAM. Also some boot functions use RAM one time and don't release it after use. Maybe I should search for the old threads, but as far as I recognize there were less than 30k available right after startup.

r44083, 2019-09-18:
@SaimenSays, According to ESP8266_RTOS_SDK latest release notes, there is 76,7 kB of free heap on v3.2 release: https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.2