SuperHouse/esp-open-rtos #705: Reading EEPROM in ESP8266

vaibhavambekar, 2019-03-31:
Hello everyone,
    I am trying to program the esp8266  with ESP-OPEN-RTOS examples .I want to store data in EEPROM  .But i have no idea about EEPROM read in ESP.because there is no reference or example available on write and read ESP-OPEN-RTOS. so i am little bit confuse  that ,can i used  EEPROM  through  this way?

I tride to add this project file but it showing lot of error .How can I do that ,so I can easily read and write in esp8266 EEPROM .

dudanov, 2019-04-08:
ESP8266 does not have EEPROM. Use filesystem in Flash Memory.

nopnop2002, 2019-04-09:
Try examples/sysparam_editor.
You can use Non-volatilestorage.

vaibhavambekar, 2019-04-12:
Thanks for reply...
  Previously I was wondering How to store data in esp8266,just like other
controller.
  But I get to knew that it's reserve space which  allowable in esp8266
using spi _flash file system.

So I get it.. Thanks once again.

vaibhavambekar, 2019-04-12:
Thanks to reply

closed 2019-04-12 by vaibhavambekar

vaibhavambekar, 2019-04-13:
sorry ,
I didn't get any ideas from above  examples/sysparam_editor.because reading and writing pattern are totally unallowable and cause some error .

Could someone help me by write and then read pattern( instruction ) for esp8266 simply to store  5 int value from fixed memory location .
 
And if i write it then, how will it affect other operation of esp8266 ?

ourairquality, 2019-04-13:
Try completely erasing the flash first. Check the flash size configured - for a nodemcu board you probably want `FLASH_SIZE = 32`. Otherwise you might have to debug what is going wrong, e.g. is the sysparam being initialized successfully etc.

nopnop2002, 2019-04-14:
When it's used for the first time.
[NOTE: No current sysparam region (initialization problem during boot?)]
There is a thing which becomes the error which is so, so please clear parameter territory by the reformat command at that time.

vaibhavambekar, 2019-04-17:
sorry for sort information regarding to my work.
I am working on esp8266-wroom-02 .which I use to read the sensor value and post on cloud using the mqtt protocol.but with the same I want to read and store sensor value in esp8266 flash  and read it on later .  

I been abled to read with the spi file in esp open rtos that it take address lower bits as the value in the example .

Before posting the previous doubt ,I wasn't get such problem with the flash memory reset but later on I figured out  that,like general firmware write there might be reset during the flashing the code(firmware) so esp8266 went to reset.and that what I am experiencing ..

previous comment suggested  me to read/write flash using  **examples/sysparam_editor** but I couldn't understand it till now.I striven to get some result but I couldn't. 

I refereed the sample code of "Basic example: read and write information from sectors on a flash memory from     [](https://www.espressif.com/sites/default/files/20a-esp8266_rtos_sdk_programming_guide_en_v1.4.0.pdf)  programming guide.
and same with  esp-open-rtos/include/espressif/spi_flash.h 

on compile and run, the read example worked correctly and even if i changed the structure according to me it also worked but when i introduced the write function inside the program it show causes problems.

nopnop2002, 2019-04-17:
https://www.espressif.com/sites/default/files/20a-esp8266_rtos_sdk_programming_guide_en_v1.4.0.pdf 

This is documento for ESP8266-OPEN-RTOS.
It's NOT ESP-OPEN-RTOS.

https://github.com/SuperHouse/esp-open-rtos/wiki/ESP-SDK- Differences

vaibhavambekar, 2019-04-19:
Thanks for quick reply..
sorry for typo about  "esp8266-wroom-02",it is Esp Wroom -02 only.