SuperHouse/esp-open-rtos #719: Support for Si7021 sensor

skanico, 2019-05-30:
As the Si7021 is now highly recommended by Sonoff, could it be possible to add support for this sensor? From other readings, it seems the protocol is very similar to DHT22 (only timing differs?). Thanks.

skanico, 2019-05-31:
Here is a [link](https://github.com/arendst/Sonoff-Tasmota/issues/735#issuecomment-348718383) where the protocol is explained, and the related [code](https://github.com/arendst/Sonoff-Tasmota/commit/253d60fd2b0c3d452d0dd8798ff767a1b143be98)

Maybe this can help :-)

mafyata, 2019-06-07:
I am interesting too. I have th16 and si7021.

skanico, 2019-06-07:
To be precise, there are 2 variants of Si7021 :
- the "original" Si7021 which is a 2-wire sensor, protocol i2c) is explained [here](https://www.silabs.com/documents/public/data-sheets/Si7021-A20.pdf)
- and "Itead Si7021", which is a modified 1-wire sensor (to make it appear like a DHT22), protocol explained above. This is the one used by Sonoff on TH16.
I think both of them could have an interest in the future given the advantages of this sensor.

RavenSystem, 2019-07-20:
Any news about this?

UncleRus, 2019-07-21:
You can test it with this PR: #725

closed 2019-07-24 by UncleRus