SuperHouse/esp-open-rtos #727: Updating Firmware - impossible

kiralikbeyin, 2019-08-05:
There is no way to update with OTA.

I tried esphttpd example to upload a bin file but it is only for esp32..

There is no solution for firmware update but only HomeACcessoryKid lib worked with under 560k file size..

https://github.com/HomeACcessoryKid/life-cycle-manager/issues/11
My firmware size is 666k

Any idea?

AndreaGreco, 2019-08-08:
At now OTA cannot work with binary major then half of flash size.

Infact bootloader let startup fist or second partition.
If your binary is more then a partition you cannot use OTA functionality.

Ideas
Try reduce size
You can try compile binary optimized for size the could improve your flash footprint and reduce of some byte.
Is possible that optimization reduce size of 3,5% not more.
Let try this way and report how many byte you save.

If this is not enough try with readelf and debug elf and a combo of options to dump simbols list and relative size order by biggest. Is. Biggest simbols all required or some simbols are unknown.
Try to optimize where is possible.

Last ideas, very complicated.
Create a separated binary is a second flash areas.
This on lunch reset ram load bss and data section and startup.
This separated binary perform OTA update.
Is only a Idea.

kiralikbeyin, 2019-08-08:
@AndreaGreco Thanks for reply.

I was using 2 step update with arduino. https://github.com/xoseperez/espurna/wiki/TwoStepUpdates

1-Trigger update with normal 666k firmware.
2-Get an 250k updater firmware.
3-250k updater will get 666k firmware
Does it work?

note that i am using https://github.com/SuperHouse/esp-open-rtos/issues/668
https://github.com/apiel/aboot firmware because only this robot.bin works with this lib..

I am on step 3 but it throws 

```
dhcp client start...
ip:192.168.0.11,mask:255.255.255.0,gw:192.168.0.1
[OTA]	HTTP client task starting
[OTA]	Image will be saved in OTA slot 1
[OTA]	Image will be saved in OTA magic 225
[OTA]	flash_offset 1404928
[OTA]	MAX_IMAGE_SIZE 1048576
Error:Ota has completed upgrade process, all ready for system software reset
Delay 1
Delay 2
Delay 3
Reset
rm match
del if0
usl
sul 0 0

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x3ffe8000, len 500, room 16
tail 4
chksum 0xf1
load 0x40100000, len 692, room 4

0x40100000: _stext at ??:?

tail 0
chksum 0xe6
csum 0xe6
Writing default boot config.
We should copy the new Rom from 0x57000 to 0x2000.
Clear rom at position 2000
SPIEraseSector at position 0x2000 sector 2
SPIEraseSector at position 0x3000 sector 3
SPIEraseSector at position 0x4000 sector 4
SPIEraseSector at position 0x5000 sector 5
SPIEraseSector at position 0x6000 sector 6
SPIEraseSector at position 0x7000 sector 7
SPIEraseSector at position 0x8000 sector 8
SPIEraseSector at position 0x9000 sector 9
SPIEraseSector at position 0xa000 sector 10
SPIEraseSector at position 0xb000 sector 11
SPIEraseSector at position 0xc000 sector 12
SPIEraseSector at position 0xd000 sector 13
SPIEraseSector at position 0xe000 sector 14
SPIEraseSector at position 0xf000 sector 15
SPIEraseSector at position 0x10000 sector 16
SPIEraseSector at position 0x11000 sector 17
SPIEraseSector at position 0x12000 sector 18
SPIEraseSector at position 0x13000 sector 19
SPIEraseSector at position 0x14000 sector 20
SPIEraseSector at position 0x15000 sector 21
SPIEraseSector at position 0x16000 sector 22
SPIEraseSector at position 0x17000 sector 23
SPIEraseSector at position 0x18000 sector 24
[... 1623 lines cut ...]
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3
```

AndreaGreco, 2019-08-08:
Try with same firmware flash blink.bin I inside of your firmware.
This confirm you that your code works.
Obviously after you need reflash with esp tools your firmware.

Let me know

kiralikbeyin, 2019-08-08:
@AndreaGreco I already flashed blink.bin it works. 

http_client_ota.c firmware is 265k

Not i am trying to find a way to put 666k ... @apiel 

aboot.h
```
//#define ROM_SIZE 0x55000 // 348160 bytes ABOOT ORGINAL
//#define ROM_SIZE 0x42000 // 270336 bytes
//#define ROM_SIZE 0x50000 // 327680 bytes
#define ROM_SIZE 0xAF000 // 716800 bytes

#define ROM_ADDR_1 0x2000
//#define ROM_ADDR_2 ROM_SIZE + ROM_ADDR_1 //ABOOT ORGINAL
//#define ROM_ADDR_2 0x42000
#define ROM_ADDR_2 0x51000
```

I got "Dowloaded image binary checsum is fallied " error so i changed Rboot_verified to 1 ...

esp-open-rtos/extras/http_client_ota/http_client_ota.c 
```
    Rboot_verified = rboot_verify_image(rboot_config.roms[slot], &boot_dimension, (const char **) &error_message);
    Rboot_verified = 1;
```

AndreaGreco, 2019-08-08:
You must store all code that is execute OTA process
All bootloader function all OTA code...
In a fixed sector that is not erased during flashing operation.

But, http_ota_client is not written with this purpose.

At now I'm over the beach.
Maybe when I back can checkout your source code if possible and try to understand exactly what are you doing.
Maybe together can found a solution.

But form here I cannot help you more.

I will back at 26/08

kiralikbeyin, 2019-08-08:
ok @AndreaGreco thanks maybe someone reads these before you come back.

I only changed 
#define ROM_SIZE 0x55000 // 348160 bytes 
to
#define ROM_SIZE 0x78000 // 491520 bytes

https://github.com/apiel/aboot/blob/master/aboot/aboot.h

453k firmware uploaded (not blink.bin)

but when i try 
#define ROM_SIZE 0xAF000 // 716800 bytes

````

dhcp client start...
ip:192.168.0.11,mask:255.255.255.0,gw:192.168.0.1
[OTA]	HTTP client task starting
[OTA]	Image will be saved in OTA slot 1
[OTA]	OTA magic 225
[OTA]	flash_offset 1773568
[OTA]	MAX_IMAGE_SIZE 1048576
rboot_verify_image: Bad second magic
Error:Ota has completed upgrade process, all ready for system software reset
Delay 1
Delay 2
Delay 3
Reset
rm match
del if0
usl
sul 0 0

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x3ffe8000, len 500, room 16
tail 4
chksum 0xf1
load 0x40100000, len 708, room 4

0x40100000: _stext at ??:?

tail 0
chksum 0x21
csum 0x21
Writing default boot config.
We should copy the new Rom from 0xb1000 to 0x2000.
Clear rom at position 2000
SPIEraseSector at position 0x2000 sector 2
SPIEraseSector at position 0x3000 sector 3
SPIEraseSector at position 0x4000 sector 4
SPIEraseSector at position 0x5000 sector 5
SPIEraseSector at position 0x6000 sector 6
SPIEraseSector at position 0x7000 sector 7
SPIEraseSector at position 0x8000 sector 8
SPIEraseSector at position 0x9000 sector 9
SPIEraseSector at position 0xa000 sector 10
SPIEraseSector at position 0xb000 sector 11
SPIEraseSector at position 0xc000 sector 12
SPIEraseSector at position 0xd000 sector 13
SPIEraseSector at position 0xe000 sector 14
SPIEraseSector at position 0xf000 sector 15
SPIEraseSector at position 0x10000 sector 16
SPIEraseSector at position 0x11000 sector 17
SPIEraseSector at position 0x12000 sector 18
SPIEraseSector at position 0x13000 sector 19
SPIEraseSector at position 0x14000 sector 20
SPIEraseSector at position 0x15000 sector 21
SPIEraseSector at position 0x16000 sector 22
[... 575 lines cut ...]
SPIEraseSector at position 0x152000 sector 338
SPIEraseSector at position 0x153000 sector 339
SPIEraseSector at position 0x154000 sector 340
SPIEraseSector at position 0x155000 sector 341
SPIEraseSector at position 0x156000 sector 342
SPIEraseSector at position 0x157000 sector 343
SPIEraseSector at position 0x158000 sector 344
SPIEraseSector at position 0x159000 sector 345
SPIEraseSector at position 0x15a000 sector 346
SPIEraseSector at position 0x15b000 sector 347
SPIEraseSector at position 0x15c000 sector 348
SPIEraseSector at position 0x15d000 sector 349
SPIEraseSector at position 0x15e000 sector 350
SPIEraseSector at position 0x15f000 sector 351
Writing default boot config.
Header size: 632064
Booting at ram 640256.
sp 0x3ffff710
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0xffffffef, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28):
````

apiel, 2019-08-10:
Can you give more details about the hardware you try to flash? Cause there is many kind of esp8266 and there is as well different options to flash your hardware depending of the size of his characteristic... For example I had lot issue for getting OTA to work on some sonoff devices and this was the reason why I was writing aboot (heavily inspired by rboot). But eventhough, I never really managed to get stable results, it was somehow working but not all the time. From what I remember, the only way to make OTA working properly was to use an arduino firmware instead of esp-open-rtos. Also using the arduino SDK will make your life much easier since there is much bigger community. Esp-open-rtos is not that's trivial and you might encounter multiple issue that are often difficult to solve. In another way, you will learn a lot as well ;-)

kiralikbeyin, 2019-08-10:
@apiel thanks for reply.

I always use nodemcu v3 (lolin) if code works i try with sonoff or other things. (qio mode works but i will try with dio also for nodemcu)

Your aboot lib works up to 450k firmware.

1-Which seems easy but dont know how to make because arduino user eboot.
I am using  666k esp-open-rtos firmware. I can use arduino OTA if i can switch back to  esp-open-rtos :
666k esp-open-rtos -> 265k arduino update -> new 667k esp-open-rtos

Two steps update :
1-Trigger update with normal 666k firmware.
2-Get an 265k updater firmware.
3-265k updater will get new 667k firmware

2-aboot modification and https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/http_client_ota/http_get.c

This code works with http_get.c example.
blink and a 450k firmware is ok..

```
#define ROM_SIZE 0xA6000 // 679936 bytes

#define ROM_ADDR_1 0x2000
#define ROM_ADDR_2 0x44000 
```

```
void clear_rom(uint32 pos) {
	uint32 end = 0x40000 ; //pos + ROM_SIZE;
	ets_printf("Clear rom at position %x\n", pos);
	for(;pos < end; pos += SECTOR_SIZE) {
		ets_printf("SPIEraseSector at position 0x%x sector %d\n", pos, pos / SECTOR_SIZE);
		if (SPIEraseSector(pos / SECTOR_SIZE) != 0) {
			ets_printf("SPIEraseSector err at position 0x%x sector %d\n", pos, pos / SECTOR_SIZE);
			return;
		}
	}
}
void clear_rom2(uint32 pos) {
	uint32 end = 0x100000;
	ets_printf("Clear rom at position %x\n", pos);
	for(;pos < end; pos += SECTOR_SIZE) {
		ets_printf("SPIEraseSector at position 0x%x sector %d\n", pos, pos / SECTOR_SIZE);
		if (SPIEraseSector(pos / SECTOR_SIZE) != 0) {
			ets_printf("SPIEraseSector err at position 0x%x sector %d\n", pos, pos / SECTOR_SIZE);
			return;
		}
	}
}
void copy_rom() {
	uint8 buffer[BUFFER_SIZE];
	uint32 pos = 0;
	uint32 end = ROM_SIZE;
	for(; pos < end; pos += BUFFER_SIZE) {
		ets_printf("Write at %d copy of %d\n", ROM_ADDR_1 + pos, ROM_ADDR_2 + pos);
		if (SPIRead(ROM_ADDR_2 + pos, buffer, BUFFER_SIZE) != 0) {
			ets_printf("read err %d\n", ROM_ADDR_2 + pos);
			return;
		}
		SPIWrite(ROM_ADDR_1 + pos, buffer, BUFFER_SIZE);
	}
}

uint32 NOINLINE find_image(void) {
	uint32 runAddr;
	uint8 rom_to_boot;

	rom_to_boot = write_conf(NO); // Sometime it reboot before to finish the copy, so lets update the current_rom only when copy is finished
	if (rom_to_boot > 0) {
		ets_printf("We should copy the new Rom from 0x%x to 0x%x.\r\n", ROM_ADDR_2, ROM_ADDR_1);
		//clear_rom(ROM_ADDR_1);
		copy_rom();
			clear_rom2(0xE4000);
//	clear_rom(ROM_ADDR_2);
		write_conf(YES);
	} else {
//	clear_rom(ROM_ADDR_2);
			clear_rom2(0xE4000);

	}
```

+ i dont want to learn more ota shit :) 10 days enough

apiel, 2019-08-12:
10 days is nothing... OTA and bootloader are not that simple.

If your code is working on nodemcu, it will not necessarily work on sonoff.

apiel, 2019-08-12:
10 days is nothing... OTA and bootloader are not that simple.

If your code is working on nodemcu, it will not necessarily work on sonoff. Some of the sonoff have very small memory size. Also some version are using 8266, some 8285... The basic sonoff, with 1 relay has different versions, with different type of memory. If I remember correctly, you have to flash them in dout.

You seem to have a problem of memory size, if I remember correctly rboot split the memory in 2 equivalent size space so this might not fit for you. But nothing stop you to customize rboot to use fix sizes. After you could try to have small ota firmware on one first boot position and then the big firmware on the second position, and keep it fix. So when you run your ota update it would always go on the second boot position. Even if theoretically this could work, it might still be buggy, because of erasing and writing issue that sometimes randomly happen. I think I was able to get some better result by erasing memory when firmware start before anything else. If you manage to get all that stuff working well would be great that you write an article and share it with us ;-)

Last thing, you should seriously think about to reduce the size of your big firmware. Be careful with this microcontroller, even if it's super powerful don't try to do to much, try to keep simple. Especially try to avoid speudo multitasking that's come with rtos, because it's not the purpose of this microcontroller.

closed 2019-08-26 by kiralikbeyin