SuperHouse/esp-open-rtos #738: Ping using ICMP ECHO_REQUEST

nonameplum, 2019-12-05:
Hi,
I'm quite a newbie in the ESP programming world.

I have searched the whole internet but couldn't find information if I can implement something similar to https://www.geeksforgeeks.org/ping-in-c/ (I don't need DNS lookup etc). The only thing I need is to check if the given IP address in local WiFi is reachable.

I implemented something similar using Ping by TCP but the downside is that there must be an open socket port on the IP that I try to PING or ping some external server e.g. http address on port 80.

nonameplum, 2019-12-07:
I have found that on ESP-IDF this could be easily achieved by https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/icmp_echo.html.

maxgerhardt, 2019-12-07:
According to the LwIP example [apps/ping](https://github.com/goertzenator/lwip/blob/master/contrib-1.4.0/apps/ping/) you can just open a raw socket and construct the ICMP echo header and data yourself. Have you tried using that code?

nonameplum, 2019-12-08:
@maxgerhardt I wasn't aware of the LwIP ping so what I have found is:
* `./esp-open-sdk/esp-open-lwip/include/lwip/app/ping.h`
* `./esp-open-sdk/esp-open-lwip/lwip/app/ping.c`

After some fight with the imports and different functions used in the `esp-open-sdk` I compiled it but I didn't run on the ESP8266 device yet. I will check it on Monday if all that work was worth it.

But anyway seems like your suggestion is far more superior as the implementation should be already compatible with `esp-open-rtos`.

You can take a look at my `ping.c` port here https://github.com/nonameplum/esp-homekit-demo/commit/60437fa101ca7eabd1faf6cdd29b360c599ebe3e

nonameplum, 2019-12-08:
@maxgerhardt Do you have an idea how to use https://github.com/SuperHouse/esp-open-rtos/issues/738#issuecomment-562867728 it? 
I have made changes to my project and I able to build it with LwIP ping implementation but I'm not sure how to check if ping was send correctly, received etc. 

In the implementation of ping in `esp-open-sdk` I can easier assume how it might work as you can explicitly specify the callbacks. But still I have no idea if it will even work on the device.

maxgerhardt, 2019-12-08:
The LwIP I referenced was that which is reference by esp-open-rtos in https://github.com/SuperHouse/esp-open-rtos/tree/master/lwip, not the one esp-open-sdk (which seems to be different?) uses. 

I'll try to quickly build a minimal **esp-open-rtos** application which demonstrates and verifies the ping functionality.

nonameplum, 2019-12-08:
Seems like in the project configuration that I have I don't have ping.h. I see that `esp-open-rtos` is pointing to https://github.com/SuperHouse/esp-open-rtos/tree/master/lwip which doesn't contain `contrip` part https://github.com/ourairquality/lwip/tree/d74e9ad2f7c9db996fb398cd41bf59ef463ae6fe

Btw:
My project is based on https://github.com/maximkulkin/esp-homekit-demo

nonameplum, 2019-12-08:
@maxgerhardt Thank you very much. I have no words to people like you.

maxgerhardt, 2019-12-09:
I got my development setup working fine and I can compile the `ping` example. I'll write a small wrapper so that it's easiert to get just the "ping result". You just want an API to ping a hostname and the result to be like ping OK, ping timeout, destination unreachable, etc. right?

nonameplum, 2019-12-09:
Yes I would like to only know if my router ip is accessible (gateway)

maxgerhardt, 2019-12-09:
Yeah the code works, I'll upload it soon with a nice one-shot ping measurement wrapper.

```
Connecting.. status 1
ip:192.168.1.135,mask:255.255.255.0,gw:192.168.1.1
Connected
Got DHCP IP: 192.168.1.135
Got GW IP: 192.168.1.1
Got Netmask: 255.255.255.0
PING thread
ping: recv 192.168.1.1 5 ms
ping: recv 192.168.1.1 2 ms
ping: recv 192.168.1.1 15 ms
ping: recv 192.168.1.1 2 ms
ping: recv 192.168.1.1 19 ms
ping: recv 192.168.1.1 12 ms
ping: recv 192.168.1.1 27 ms
```

nonameplum, 2019-12-09:
Wow. Sounds perfect

maxgerhardt, 2019-12-10:
@nonameplum Hi, could you try this example please: https://github.com/maxgerhardt/esp-open-rtos-ping-example

nonameplum, 2019-12-15:
Works perfect. Thanks again.

closed 2019-12-15 by nonameplum