SuperHouse/esp-open-rtos #739: Print hex commands - QUESTION

HomeHomeApp, 2020-01-29:
This is a working arduino sketch:

````
byte relON[] = {0xA0, 0x01, 0x01, 0xA2};  //Hex command to send to serial for open relay
byte relOFF[] = {0xA0, 0x01, 0x00, 0xA1}; //Hex command to send to serial for close relay

int ledState = false;
unsigned long previousMillis = 0;
const long interval = 2000; //  2 seconds

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize serial:
  Serial.begin(9600);
}

// the loop function runs over and over again forever
void loop()
{
  unsigned long currentMillis = millis();
  if(currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;   
    if (ledState == true) {
      Serial.write(relON, sizeof(relON));     // turns the relay ON
      ledState = false;
    } else {
      Serial.write(relOFF, sizeof(relOFF));   // turns the relay OFF
      ledState = true;
    }    
  }
}
`````
How can i print hex commands with esp-open-rtos?

(lctech device : http://www.chinalctech.com/cpzx/1/333.html)

problemss, 2020-03-10:
I have this same exact question and similar board. Did you ever figure it out?

kiralikbeyin, 2020-03-10:
@problemss 
Try this:
````
 byte relON[] = {160,1,1,162};  //Hex command to send to serial for open relay
 byte relOFF[] = {160,1,0,161}; //Hex command to send to serial for close relay

 char relON[] = {160,1,1,162};  //Hex command to send to serial for open relay
 char relOFF[] = {160,1,0,161}; //Hex command to send to serial for close relay

 char relONhex[] = {0xA0,0x01,0x01,0xA2};
 char relOFFhex[] = {0xA0,0x01,0x00,0xA1};

void user_init(void) {

...
      if(on)
      {
        //printf("%.*s\r\n", (int)sizeof(relONhex), relONhex);
        printf("%.*s\n", (int)sizeof(relON), relON);
      }
      else
      {
        //printf("%.*s\r\n", (int)sizeof(relOFFhex), relOFFhex);
        printf("%.*s\n", (int)sizeof(relOFF), relOFF);

      }
 }
````
It will turn on but i tried everything, it will turn off only in the same code block.

Put it to your main code block after serial start.
wait 3 seconds
turn on
wait 3 seconds
turn off

**Somehow when it is on, you can only turn off it in the same code block. This maked me sick!**

When code block comes to end it will not work again (i tried while(1) {...} and also as a seperate function )

Please write if you can find a solution.

problemss, 2020-03-12:
@kiralikbeyin I found a solution!!!

Man I tried so many things, but the cause is so simple. when you pass a 0 it is treated as a null. When printing strings, the null character is used to designate the end of the string. So printf is only printing the first two characters and then stopping. To fix this you have to print the 0 by itself.

```
printf("%.*s", (int)sizeof(c_relOFF1),c_relOFF1);
printf("%c",'\0');
printf("%.*s\n", (int)sizeof(c_relOFF3),c_relOFF3);
```

Where the varibles are defined as
```
char c_relOFF1[] = {160, 1};
char c_relOFF3[] = {161};
```

Ignore my variable numbering as I have been trying so many things, I will be going back in and cleaning up the code now that it is working!

kiralikbeyin, 2020-03-13:
@problemss works fine thank you!