SuperHouse/esp-open-rtos #761: Approach to sniff ESP32 packets using Wireshark

Osama-Ramadan, 2020-08-30:
I am using ESP32 to send some data from the esp to some server. 
My laptop is connected to the same router as my esp and the router is using WPA2 as security. With my little reading, I understand that under such security wifi clients will not be able to see the traffic of each other. and therefore I am not able to see the packets through Wireshark on my laptop. 

**So the question: is there exists a way that allows me to be able to sniff the packets sent from the esp to the server using Wireshark on my laptop?**

I hope I was able to explain my question in an obvious way 
Thank you for your help

anszom, 2020-09-02:
Your question doesn't seem to relate to this project in any way... but anyway here is a guide to decrypting WPA traffic with wireshark: https://wiki.wireshark.org/HowToDecrypt802.11 - it should work with ESP32 or any other wifi station.

Osama-Ramadan, 2020-09-03:
@anszom Thank you so much for your replay. 
I am sorry what happened is I was searching for this problem and found another issue with a similar question here, so I thought this is the right place to place my question.