SuperHouse/esp-open-rtos #762: pcf8574 not working

QDUNI, 2020-09-22:
Hi, I have an esp8266ex and a pcf8574at. 
Now im trying to use the pcf8574 library but with no succes. 
can you show me how I need to use it? 

addr = 0x38

I want all the ports set to output so I can write them on and off. 
I started with initializing every pin as output and then turn them all off. but with no succes. 

   `pcf8574_gpio_write(&dev, 0, false);
    pcf8574_gpio_write(&dev, 1, false);
    pcf8574_gpio_write(&dev, 2, false);
    pcf8574_gpio_write(&dev, 3, false);
    pcf8574_gpio_write(&dev, 4, false);
    pcf8574_gpio_write(&dev, 5, false);
    pcf8574_gpio_write(&dev, 6, false);
    pcf8574_gpio_write(&dev, 7, false;
    pcf8574_port_write(&dev, 0);`

rangelflavio, 2020-10-19:
enable debug on i2c driver - make a loop to view and try each write address - some ic's come with another address

QDUNI, 2020-11-06:
Do you have an example ?

QDUNI, 2020-11-06:
If I use the Arduino library the address is 0x38... Im also able to use the pcf8574 with Arduino.
I want to make it work with esp open rtos, because of homekit