SuperHouse/esp-open-rtos #763: lvgl/lv_drivers -- fatal: reference is not a tree

alejandrogzu, 2020-10-03:
Hello!

i am trying to clone the repository but I get this error, I am building in Ubuntu 16.04, this is the docker file:

---

FROM ubuntu:16.04 as builder
RUN  apt-get update && apt-get install -y git
RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos

----

and I get this:

docker build . -f esp-rtos-dockerfile -t esp-rtos
[+] Building 40.9s (7/9)                                                                                                                                           
 => [internal] load build definition from esp-rtos-dockerfile                                                                                                 0.0s
 => => transferring dockerfile: 822B                                                                                                                          0.0s
 => [internal] load .dockerignore                                                                                                                             0.0s
 => => transferring context: 2B                                                                                                                               0.0s
 => [internal] load metadata for docker.io/library/ubuntu:16.04                                                                                               1.2s
 => [internal] load metadata for docker.io/library/esp-sdk:latest                                                                                             0.0s
 => [builder 1/3] FROM docker.io/library/ubuntu:16.04@sha256:185fec2d6dbe9165f35e4a1136b4cf09363b328d4f850695393ca191aa1475fd                                 0.0s
 => [stage-1 1/2] FROM docker.io/library/esp-sdk:latest                                                                                                       0.0s
 => CACHED [builder 2/3] RUN apt-get update && apt-get install -y git                                                                                         0.0s
 => [builder 3/3] RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos                                               39.6s
 => => # Submodule path 'extras/multipwm': checked out '44ecea5757bee0112c65f33eadb44e91a8923535'                                                                 
 => => # Cloning into 'extras/spiffs/spiffs'...                                                                                                                   
 => => # Submodule path 'extras/spiffs/spiffs': checked out '39937743fbbec4b82308ee08332bf9180408d23b'                                                            
 => => # Cloning into 'lvgl/lv_drivers'...                                                                                                                        
 => => # fatal: reference is not a tree: 8fe0804ceb875193a37106a2e25d07c106650a86                                                                                 
 => => # Cloning into 'lvgl/lv_examples'...  

there is any solution????

i also use ubuntu 18.04 and its works but I need to use 16.04

Thanks!

PhilippvK, 2020-10-04:
Same issue here.

alejandrogzu, 2020-10-08:
Hello @PhilippvK, 

as a workaround I fork the project SuperHouse/esp-open-rtos and I delete the folders LVGL from the project, I also delete the the references of this drives in .gitmodules. you can use my repository alejandrogzu/esp-open-rtos to work, it works if you don need lvl drivers.

@maximkulkin do you have any idea what could be happening?

Thank you!!!

PhilippvK, 2020-10-08:
@alejandrogzu I found out that the version of Git coming with Ubuntu 16.04 is very outdated. Installing the latest git version resolved the Issue for me.

However, your workaround sounds like a good alternative approach.

fduman, 2020-11-03:
If you set to Ubuntu v20.04 as "FROM ubuntu:20.04 as builder" in your docker file, it works perfectly.