SuperHouse/esp-open-rtos #776: when making esp-open-sdk，cannot download isl-0.14

ye-zhang0720, 2021-11-13:
[DEBUG]    Already have 'mpfr-3.1.3'
[EXTRA]    Retrieving 'isl-0.14'
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14.tar.xz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.tar.xz.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14.tar.xz' 
[ALL  ]    --2021-11-14 00:35:14--  http://isl.gforge.inria.fr/isl-0.14.tar.xz
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:17 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14.tar.xz"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14.tar.lzma'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.tar.lzma.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14.tar.lzma' 
[ALL  ]    --2021-11-14 00:35:17--  http://isl.gforge.inria.fr/isl-0.14.tar.lzma
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:21 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14.tar.lzma"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14.tar.bz2'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.tar.bz2.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14.tar.bz2' 
[ALL  ]    --2021-11-14 00:35:21--  http://isl.gforge.inria.fr/isl-0.14.tar.bz2
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:28 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14.tar.bz2"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14.tar.gz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.tar.gz.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14.tar.gz' 
[ALL  ]    --2021-11-14 00:35:28--  http://isl.gforge.inria.fr/isl-0.14.tar.gz
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:31 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14.tar.gz"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14.tgz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.tgz.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14.tgz' 
[ALL  ]    --2021-11-14 00:35:31--  http://isl.gforge.inria.fr/isl-0.14.tgz
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:34 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14.tgz"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14.tar'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.tar.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14.tar' 
[ALL  ]    --2021-11-14 00:35:34--  http://isl.gforge.inria.fr/isl-0.14.tar
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:38 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14.tar"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14.zip'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.zip.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14.zip' 
[ALL  ]    --2021-11-14 00:35:38--  http://isl.gforge.inria.fr/isl-0.14.zip
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:41 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14.zip"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.14'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/isl-0.14.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.14' 
[ALL  ]    --2021-11-14 00:35:42--  http://isl.gforge.inria.fr/isl-0.14
[ALL  ]    Connecting to 127.0.0.1:7890... connected.
[ALL  ]    Proxy request sent, awaiting response... 502 Bad Gateway
[ALL  ]    2021-11-14 00:35:48 ERROR 502: Bad Gateway.
[ALL  ]    
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14"
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_isl_get[scripts/build/companion_libs/121-isl.sh@16]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@591]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 0:34.73)

thirstyone, 2022-01-18:
just download that file from a mirror and put it into /home/zy/Downloads/esp/esp-open-sdk/crosstool-NG/.build/tarballs/ directory

ye-zhang0720, 2022-01-19:
Thank you so much 

张烨
Zhang Ye
***@***.***
            ***@***.***