SuperHouse/esp-open-rtos #777: BearSSL SSL certificate has expired

thirstyone, 2022-01-18:
when doing git clone --recursive https://github.com/Superhouse/esp-open-rtos.git I get the following error: 
Cloning into '/Volumes/case-sensitive/esp-open-sdk/esp-open-rtos/extras/bearssl/BearSSL'...
fatal: unable to access 'https://www.bearssl.org/git/BearSSL/': SSL certificate problem: certificate has expired

nochkin, 2022-01-19:
The cert is good and valid until Feb 24th. You may want to check the local time/date on your host.

thirstyone, 2022-01-19:
I know, I've checked both. That's weird. 
date "+DATE: %Y-%m-%d%nTIME: %H:%M:%S"
DATE: 2022-01-19
TIME: 03:22:57

MITM?

nochkin, 2022-01-19:
I don't think it's MITM, must be something silly.
Here is what I see when I query for the cert:

```
$ echo Q | openssl s_client -connect www.bearssl.org:443 | openssl x509 -noout -dates
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = bearssl.org
verify return:1
DONE
notBefore=Nov 26 07:56:32 2021 GMT
notAfter=Feb 24 07:56:31 2022 GMT

```
What do you see when you execute the same command? I wonder if "has expired" is just an incorrect error message.

thirstyone, 2022-01-19:
Tried it here: 
echo Q | openssl s_client -connect www.bearssl.org:443 | openssl x509 -noout -dates
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify error:num=20:unable to get local issuer certificate
verify return:0
DONE
notBefore=Nov 26 07:56:32 2021 GMT
notAfter=Feb 24 07:56:31 2022 GMT

nochkin, 2022-01-19:
The dates are good, it's your chain which could not be verified (see your "verify error" message).
I suspect this is your CA Root got expired or missing. Check this one: https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

thirstyone, 2022-01-19:
verify returned 0 instead of 1. 
do you suggest upgrading OS on my obsolete macbook?

nochkin, 2022-01-19:
If upgrading OS is an option, that would do it.
If you can't do it for any reason, get the proper CA cert into your KeyChain on OSX: https://www.tenseg.net/blog/2021/10/10/installing-the-new-lets-encrypt-root-certificate/

thirstyone, 2022-01-19:
thanks, but it didn't work - probably, a different keychain (and a different certificate chain?) is used by browser and wget (or whatever) - that would explain that my browser initially has no problems with opening that website. 
is there another workaround, like cloning that git manually?

thirstyone, 2022-01-19:
btw, when I open https://www.bearssl.org/git/BearSSL in chrome, I seee 404 error

nochkin, 2022-01-19:
Getting 404 is ok as you are not accessing Git repo in your browser. But at least your updated CA root is readable by your browser. Now you need to make it working using Git client.
Try this under sudo/root on your OSX:
```
% cd /etc/ssl
% sudo mv cert.pem cert.pem.orig
% sudo wget -O cert.pem https://curl.se/ca/cacert.pem
```
That will make a backup copy of your original cert.pem file and grab online one of the latest ones.

As a verification, you can do either one:
`% echo Q | openssl s_client -connect www.bearssl.org:443 | openssl x509 -noout -dates`
You should see "verify return:1" and no errors.
or
`% git clone https://www.bearssl.org/git/BearSSL`
You should get Git repo cloned successfully.

thirstyone, 2022-01-19:
once again: I had no issues accessing that site with my browser. 
after these commands, I still get the same error: `echo Q | openssl s_client -connect www.bearssl.org:443 | openssl x509 -noout -dates
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify error:num=20:unable to get local issuer certificate
verify return:0
DONE
notBefore=Nov 26 07:56:32 2021 GMT
notAfter=Feb 24 07:56:31 2022 GMT`

old version of openssl? 
`openssl version -a
LibreSSL 2.2.7
built on: date not available
platform: information not available
options:  bn(64,64) rc4(ptr,int) des(idx,cisc,16,int) blowfish(idx) 
compiler: information not available
OPENSSLDIR: "/private/etc/ssl"`